platformio/platformio-core-installer #2380: Linux:

BernH4, 2022-08-20:
The install fails because it cant find python in the specified path: 
`No such file or directory: '/home/bs/.platformio/penv/bin/python'`

It seems to search in the penv/bin folder but the script but the symlinks or bin files in another directory:
penv/**local**/bin

```
➜  ~ python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
Installer version: 1.1.2
Platform: Linux-5.15.0-46-generic-x86_64-with-glibc2.35
Python version: 3.10.4 (main, Jun 29 2022, 12:14:53) [GCC 11.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/bs/.platformio/penv
Traceback (most recent call last):
  File "/tmp/tmpv8smpcvy", line 68, in <module>
    main()
  File "/tmp/tmpv8smpcvy", line 60, in main
    bootstrap()
  File "/tmp/tmpv8smpcvy", line 46, in bootstrap
    pioinstaller.__main__.main()
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/click/core.py", line 1053, in main
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/tmp/.piocore-installer-7gn4xpp7/tmp0zzqmvwa/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/bs/.platformio/penv/bin/python'
Traceback (most recent call last):
  File "<string>", line 105, in <module>
  File "<string>", line 101, in main
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '/tmp/tmpv8smpcvy']' returned non-zero exit status 1.
➜  ~ 
```

zingo, 2022-09-18:
Could be the same problem as https://github.com/platformio/platformio-core-installer/issues/2404

zingo, 2022-09-20:
Installing python3-venv as mentioned in https://github.com/platformio/platformio-core-installer/issues/85
solved it for me.

closed 2022-11-01 by stale[bot]