platformio/platformio-core-installer #2402: PIO Core Call Error: "[WinError 2] The system cannot find the file specified"

wonderingStars, 2022-09-01:
What kind of issue is this?
PIO Core Call Error: "[WinError 2] The system cannot find the file specified"
-

------------------------------------------------------------------

### Configuration
intel 10900
32gb ram
**Operating system**:
Windows 11

### Description of problem
**PlatformIO Version** (`platformio --version`):
VScode with platfrom.io IDE plugin

Insert here...
Could not initialize project
PIO Core Call Error: "[WinError 2] The system cannot find the file specified"

### Additional info
i have reinstalled the plugin restated the machine 
i have uninstall VS code  and the plugin restarted the machine reinstalled them still nothing works .

ivankravets, 2022-09-01:
Could you remove `C:/Users/%username%/.platformio/penv` folder and restart VSCode?

wonderingStars, 2022-09-01:
I got it working with Clion now so I will close this .

closed 2022-09-01 by ivankravets