platformio/platformio-core-installer #2404: Installation Manager

Freireg, 2022-09-03:
%23 Description of problem
I'm trying to install the VSCode extension on Ubuntu 22 and it doesn't finish the installation process, returning the error below. I tried re-instaling it after deleting any related folder. After following the installations first step on the oficiall website. 
I saw some similar but outdated posts, any idea of what might be?

  %23 Configuration

  VSCode: 1.71.0
  PIO IDE: v2.5.2
  System: Linux, 5.15.0-47-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/guilherme/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/home/guilherme/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/home/guilherme/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/guilherme/.platformio/.cache/tmp/.piocore-installer-w31l13ye/tmpqs3xumyn/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/guilherme/.platformio/penv/bin/python'
 -> Installer version: 1.1.2
Platform: Linux-5.15.0-47-generic-x86_64-with-glibc2.35
Python version: 3.10.4 (main, Jun 29 2022, 12:14:53) [GCC 11.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/guilherme/.platformio/penv

	at /home/guilherme/.vscode/extensions/platformio.platformio-ide-2.5.2-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5308
	at ChildProcess.o (/home/guilherme/.vscode/extensions/platformio.platformio-ide-2.5.2-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4691)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

zingo, 2022-09-18:
I see the same, also tried to delete folders. I had a flatpack version installed before so I thought it might have caused the problem but I see your report here and https://github.com/platformio/platformio-core-installer/issues/2433 that seem to be the same.

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.67.2
  PIO IDE: v2.5.4
  System: Linux, 5.19.0-76051900-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/zingo/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/home/zingo/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/home/zingo/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/zingo/.platformio/.cache/tmp/.piocore-installer-zore82q5/tmp2kisnyfw/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/zingo/.platformio/penv/bin/python'
 -> Installer version: 1.1.2
Platform: Linux-5.19.0-76051900-generic-x86_64-with-glibc2.35
Python version: 3.10.4 (main, Jun 29 2022, 12:14:53) [GCC 11.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/zingo/.platformio/penv

	at /home/zingo/.vscode/extensions/platformio.platformio-ide-2.5.4-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5308
	at ChildProcess.o (/home/zingo/.vscode/extensions/platformio.platformio-ide-2.5.4-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4691)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

zingo, 2022-09-18:
While browsing the other issues with the same problem I wonder if there could be some mixup or something with this two folders

- /home/zingo/.platformio/penv/local/bin
- /home/zingo/.platformio/penv/bin

`/home/zingo/.platformio/penv/local/bin` contan python but not `/home/zingo/.platformio/penv/bin` that is later used?

Freireg, 2022-09-18:
Some days after publishing this issue the platformio was working properly. Don't know what resolved this installation manager issue, maybe my Ubuntu just needed a specific update

closed 2022-09-18 by Freireg

zingo, 2022-09-20:
Installing python3-venv as mentioned in https://github.com/platformio/platformio-core-installer/issues/85
solved it for me on popOS! based on Ubuntu 22.04.