platformio/platformio-core-installer #2433: Installation Manager

teebrin, 2022-09-16:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.71.1
  PIO IDE: v2.5.4
  System: Darwin, 21.6.0, arm64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/Users/yl/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/Users/yl/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/Users/yl/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/click/core.py", line 1053, in main
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/Users/yl/.platformio/.cache/tmp/.piocore-installer-38dqtmef/tmp1jpjqzd6/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/Users/yl/homebrew/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/Users/yl/homebrew/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Users/yl/homebrew/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Users/yl/homebrew/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/Users/yl/.platformio/penv/bin/python'
 -> Installer version: 1.1.2
Platform: macOS-12.6
Python version: 3.10.6 (main, Aug 18 2022, 03:09:00) [Clang 13.1.6 (clang-1316.0.21.2.5)]
Python path: /Users/yl/homebrew/opt/python@3.10/bin/python3.10
Creating a virtual environment at /Users/yl/.platformio/penv

	at /Users/yl/.vscode/extensions/platformio.platformio-ide-2.5.4-darwin-arm64/node_modules/platformio-node-helpers/dist/index.js:1:5308
	at ChildProcess.o (/Users/yl/.vscode/extensions/platformio.platformio-ide-2.5.4-darwin-arm64/node_modules/platformio-node-helpers/dist/index.js:1:4691)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at Socket.<anonymous> (node:internal/child_process:451:11)
	at Socket.emit (node:events:526:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

zingo, 2022-09-18:
Could be the same problem as https://github.com/platformio/platformio-core-installer/issues/2404

zingo, 2022-09-20:
https://github.com/platformio/platformio-core-installer/issues/85
solved it for me

closed 2022-11-01 by stale[bot]