platformio/platformio-core-installer #2582: Installation Manager

dzalf, 2022-11-19:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.73.1
  PIO IDE: v2.5.5
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Error: Compatible PlatformIO Core not found.
Reason: Could not run `C:\Users\dmele\.platformio\penv\Scripts\platformio.exe --help`.
Error: Error: Traceback (most recent call last):

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main

    cli()  %23 pylint: disable=no-value-for-parameter

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__

    return self.main(*args, **kwargs)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1054, in main

    with self.make_context(prog_name, args, **extra) as ctx:

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 920, in make_context

    self.parse_args(ctx, args)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1613, in parse_args

    rest = super().parse_args(ctx, args)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1378, in parse_args

    value, args = param.handle_parse_result(ctx, opts, args)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 2360, in handle_parse_result

    value = self.process_value(ctx, value)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 2322, in process_value

    value = self.callback(ctx, self, value)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1273, in show_help

    echo(ctx.get_help(), color=ctx.color)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 699, in get_help

    return self.command.get_help(self)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1298, in get_help

    self.format_help(ctx, formatter)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1331, in format_help

    self.format_options(ctx, formatter)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1533, in format_options

    self.format_commands(ctx, formatter)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\click\core.py", line 1586, in format_commands

    cmd = self.get_command(ctx, subcommand)

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\platformio\cli.py", line 80, in get_command

    module = importlib.import_module(commands[cmd_name])

  File "C:\Users\dmele\.platformio\python3\lib\importlib\__init__.py", line 127, in import_module

    return _bootstrap._gcd_import(name[level:], package, level)

  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import

  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load

  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked

  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked

  File "<frozen importlib._bootstrap_external>", line 850, in exec_module

  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\platformio\commands\ci.py", line 24, in <module>

    from platformio.project.commands.init import project_init_cmd, validate_boards

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\platformio\project\commands\init.py", line 55, in <module>

    @click.option("--ide", type=click.Choice(ProjectGenerator.get_supported_ides()))

  File "C:\Users\dmele\.platformio\penv\lib\site-packages\platformio\project\integration\generator.py", line 62, in get_supported_ides

    for name in os.listdir(tpls_dir)

FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\dmele\\.platformio\\penv\\lib\\site-packages\\platformio\\assets\\templates\\ide-projects'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using

  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section

  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers

  https://github.com/platformio/platformio-core/issues

============================================================

    at env (c:\Users\dmele\.vscode\extensions\platformio.platformio-ide-2.5.5-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:217:25)
    at ChildProcess.o (c:\Users\dmele\.vscode\extensions\platformio.platformio-ide-2.5.5-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:165:74)
    at ChildProcess.emit (node:events:526:28)
    at ChildProcess.emit (node:domain:475:12)
    at maybeClose (node:internal/child_process:1092:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

dzalf, 2022-11-19:
I have uninstalled the extension 3 times, deleted the .platformio folder and NOTHING. The extension is broken and I have a HUGE development I need to work with over the weekend >:(

[image]

[image]

dzalf, 2022-11-19:
I followed the recommendation regarding installing the latest version of pio:

[image]

oliver-eifler, 2022-11-19:
have had the same issue
after additionl install of platformio core with get-platformio.py script it works again
https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#local-download-macos-linux-windows

dzalf, 2022-11-21:
> have had the same issue after additionl install of platformio core with get-platformio.py script it works again https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#local-download-macos-linux-windows

Dear Oliver

Thank you for your input. 

I will try to test your solution as soon as I can. To do so I will have to "break" my platformio installation _again_ since I might have found a way to reproduce my problem and now it is fixed in a way.

It turns out that after several failed attempts, I decided to wipe VSCode and all its related folders including the .platformio one.
The trick was to **NOT SYNC** my settings. Understandably, I work across multiple computers and I used to have all my settings synced (extensions, VSCode behaviour, themes, etc).

When reinstalling all my extensions from scratch, platformio would work flawlessly, HOWEVER, as soon as I activated syncing in VSCode everything went haywire upon restarting VSCode where pio attempted reinstalling and failing multiple times. This is an extremely weird behaviour that had never happened to me before.

I decided to delete all my sync data and start with a clean cloud set of settings, based on the ones that worked at last (the brand-new installation on my main PC).

Perhaps this is a bug that the developers might want to look into: syncing settings on VSCode breaks the pio installation (@ivankravets). I have attached a copy of my previous settings.json (in txt extension) file in case it helps trace down the origin of my issue: [settings.txt](https://github.com/platformio/platformio-core-installer/files/10056575/settings.txt)

Thank you, guys!

Cheers 🍻

ivankravets, 2022-11-25:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2620

closed 2022-11-25 by ivankravets