platformio/platformio-core-installer #2851: Installation Manager

Mejituu, 2023-04-09:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.1
  PIO IDE: v3.1.1
  System: Linux, 6.1.0-7-amd64, x64

  %23 Exception
  ```
  Error: /home/mejituu/.platformio/.cache/tmp/.piocore-installer-gz5kqg_0/tmpx_lr1yrk/pioinstaller.zip/requests/__init__.py:102: RequestsDependencyWarning: urllib3 (1.26.9) or chardet (5.1.0)/charset_normalizer (2.0.12) doesn't match a supported version!
Error: Compatible PlatformIO Core not found.
Reason: PlatformIO Core version 6.0.1 does not match version requirements >=6.1.6.

	at /home/mejituu/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5303
	at ChildProcess.o (/home/mejituu/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at Socket.<anonymous> (node:internal/child_process:451:11)
	at Socket.emit (node:events:526:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

ivankravets, 2023-05-10:
Please uninstall global PlatformIO Cores https://docs.platformio.org/en/latest/core/installation/troubleshooting.html#multiple-platformio-cores-in-a-system

closed 2023-05-10 by ivankravets

mhavill, 2024-01-21:
I decided to update PIO extension to v3.3.2 today,  but I cannot get past this error:
```
Error: Error: Compatible PlatformIO Core not found. 
Reason: Could not run `C:\Users\mick\.platformio\penv\Scripts\platformio.exe --help`. 
Error: ImportError: Traceback (most recent call last): 
File "C:\Users\mick\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main cli() # pylint: disable=no-value-for-parameter ^^^^^ 
File "C:\Users\mick\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__ return self.main(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^ 
File "C:\Users\mick\.platformio\penv\Lib\site-packages\click\core.py", line 1077, in main with self.make_context(prog_name, args, **extra) as ctx: ...
```
I have followed the links and this solution including deleting .platformio directory completely. 
W10 
AV disabled
Very odd!  After a clean install of the extension when I run  pip uninstall platformio
I get the message

```
Found existing installation: platformio **6.1.13**
Uninstalling platformio-6.1.13:
  Would remove:
    c:\users\mick\.platformio\penv\lib\site-packages\platformio-6.1.13.dist-info\*
    c:\users\mick\.platformio\penv\lib\site-packages\platformio\*
    c:\users\mick\.platformio\penv\scripts\pio.exe
    c:\users\mick\.platformio\penv\scripts\piodebuggdb.exe
    c:\users\mick\.platformio\penv\scripts\platformio.exe
Proceed (Y/n)? Y
```
How can it be installing an old version?

mhavill, 2024-01-21:
I found my problem was an older version of Python was being used even though I had updated - multiple copies.  I removed Python310 leaving Python 311

**As always, thanks Max and Ivan!**

ianzone, 2024-12-17:
The built-in pio always fails to download, there should be a fallback solution to use the global pio.

jonahrakotoarijaona-dot, 2025-08-13:
Error: Could not start PIO Home server: Timeout error at Timeout._onTimeout (c:\Users\WINDOWS 10\.vscode\extensions\platformio.platformio-ide-3.3.4-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10741) at listOnTimeout (node:internal/timers:588:17) at process.processTimers (node:internal/timers:523:7)...

ramiss, 2025-12-26:
This is getting on my last nerve.  Seems like every other time I open VS Code I get this issue and have to uninstall and re-install PlatformIO.

Then sometimes I get this:

> Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/pioarduino/pioarduino-core-installer/issues at c:\Users\Richard\.vscode\extensions\pioarduino.pioarduino-ide-1.1.5\node_modules\pioarduino-node-helpers\dist\index.js:1:5093 at ChildProcess.v (c:\Users\Richard\.vscode\extensions\pioarduino.pioarduino-ide-1.1.5\node_modules\pioarduino-node-helpers\dist\index.js:1:4399) at ChildProcess.emit (node:events:519:28) at ChildProcess.emit (node:domain:489:12) at maybeClose (node:internal/child_process:1101:16) at ChildProcess._handle.onexit (node:internal/child_process:304:5)...