platformio/platformio-core-installer #2857: Installation Manager

chatta2539, 2023-04-12:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.2
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\raehchester\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "C:\Users\raehchester\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "C:\Users\raehchester\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\click\core.py", line 1637, in invoke
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\pioinstaller\core.py", line 70, in install_platformio_core
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\pioinstaller\core.py", line 98, in _install_platformio_core
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\pioinstaller\penv.py", line 71, in create_core_penv
  File "C:\Users\raehchester\.platformio\.cache\tmp\.piocore-installer-w02ah_oy\tmpgq16dslx\pioinstaller.zip\pioinstaller\penv.py", line 141, in init_state
  File "C:\Users\raehchester\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\raehchester\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\raehchester\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\raehchester\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\raehchester\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\raehchester\.platformio\penv

    at spawnOptions (c:\Users\raehchester\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:226:25)
    at ChildProcess.o (c:\Users\raehchester\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:170:13)
    at ChildProcess.emit (node:events:526:28)
    at ChildProcess.emit (node:domain:475:12)
    at maybeClose (node:internal/child_process:1092:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/884

closed 2023-05-31 by ivankravets