platformio/platformio-core-installer #2859: Installation Manager

jvmoraiscb, 2023-04-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.2
  PIO IDE: v3.1.1
  System: Linux, 5.4.0-146-generic, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Linux-5.4.0-146-generic-x86_64-with-glibc2.25
Python version: 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/breno/.platformio/penv
/usr/bin/python3

	at /home/breno/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5303
	at ChildProcess.o (/home/breno/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at Socket.<anonymous> (node:internal/child_process:451:11)
	at Socket.emit (node:events:526:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

closed 2023-04-14 by jvmoraiscb

Jie-Huangi, 2023-04-16:
can you share your ideals. My Platformio is also not work on Linux. Thank you.

jvmoraiscb, 2023-04-24:
i installed python virtualenv lib:
```bash
pip install virtualenv
```