platformio/platformio-core-installer #2870: Installation Manager

haplik68-cpu, 2023-04-20:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.3
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Haplik\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "C:\Users\Haplik\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "C:\Users\Haplik\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\pioinstaller\__main__.py", line 113, in core_check
  File "C:\Users\Haplik\.platformio\.cache\tmp\.piocore-installer-hqsq3s4a\tmprvytqk5h\pioinstaller.zip\pioinstaller\core.py", line 169, in check
  File "C:\Users\Haplik\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Haplik\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Haplik\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Haplik\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
PermissionError: [WinError 5] P��stup byl odep�en

	at c:\Users\Haplik\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (c:\Users\Haplik\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/884

closed 2023-05-31 by ivankravets