platformio/platformio-core-installer #2872: Installation Manager

custodio1d, 2023-04-21:
Installation Manager

Error: Traceback (most recent call last): File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\pioinstaller\python.py", line 170, in find_compatible_pythons File "C:\Users\xpg.platformio\python3\lib\subprocess.py", line 424, in check_output return run(*popenargs, stdout=PIPE, timeout=timeout, check=True, File "C:\Users\xpg.platformio\python3\lib\subprocess.py", line 528, in run raise CalledProcessError(retcode, process.args, subprocess.CalledProcessError: Command '['C:\Users\xpg\AppData\Local\Microsoft\WindowsApps\python3.exe', 'C:\Users\xpg\.platformio\.cache\tmp\get-platformio-0.3.5.py', '--no-shutdown-piohome', 'check', 'python']' returned non-zero exit status 9009.

During handling of the above exception, another exception occurred:

Traceback (most recent call last): File "C:\Users\xpg.platformio.cache\tmp\get-platformio-0.3.5.py", line 69, in main() File "C:\Users\xpg.platformio.cache\tmp\get-platformio-0.3.5.py", line 61, in main bootstrap() File "C:\Users\xpg.platformio.cache\tmp\get-platformio-0.3.5.py", line 47, in bootstrap pioinstaller.main.main() File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\pioinstaller__main__.py", line 122, in main File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\click\core.py", line 829, in call File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\click\core.py", line 782, in main File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\click\core.py", line 1236, in invoke File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\click\core.py", line 1066, in invoke File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\click\core.py", line 610, in invoke File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\click\decorators.py", line 21, in new_func File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\pioinstaller__main__.py", line 52, in cli File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\pioinstaller\core.py", line 72, in install_platformio_core File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\pioinstaller\penv.py", line 52, in create_core_penv File "C:\Users\xpg.platformio.cache\tmp.piocore-installer-lzsjfs32\tmplnbrn0qx\pioinstaller.zip\pioinstaller\python.py", line 191, in find_compatible_pythons UnboundLocalError: local variable 'error' referenced before assignment -> Installer version: 0.3.5 Platform: Windows-10 Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)] Python path: C:\Users\xpg.platformio\python3\python.exe Creating a virtual environment at C:\Users\xpg.platformio\penv

at C:\Users\xpg\.pulsar\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:5004
at ChildProcess.s (C:\Users\xpg\.pulsar\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:4900)
at ChildProcess.emit (events.js:315:20)
at ChildProcess.cp.emit (C:\Users\xpg\.pulsar\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
at maybeClose (internal/child_process.js:1048:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)

ivankravets, 2023-05-10:
Have you resolved this issue?

ivankravets, 2023-05-31:
Please upgrade VSCode and PlatformIO IDE for VSCode to the latest versions.

closed 2023-05-31 by ivankravets