platformio/platformio-core-installer #2873: Installation Manager

demond2633, 2023-04-22:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.3
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error: Error: Compatible PlatformIO Core not found.
Reason: PlatformIO Core version 4.0.0 does not match version requirements >=6.1.6.

	at c:\Users\sam_z\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (c:\Users\sam_z\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

chamemilo, 2023-04-22:
Hello, from what i understand you need to install a python version that is **Python 3.6+ or above.**

ivankravets, 2023-05-10:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2851

closed 2023-05-10 by ivankravets