platformio/platformio-core-installer #2876: Installation Manager

MUHAN11-c, 2023-04-26:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.73.1
  PIO IDE: v3.1.1
  System: Linux, 5.4.0-84-generic, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Linux-5.4.0-84-generic-x86_64-with-glibc2.25
Python version: 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/w/.platformio/penv
/usr/bin/python3

	at /home/w/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5303
	at ChildProcess.o (/home/w/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

MUHAN11-c, 2023-04-26:
pip install virtualenv

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2955

closed 2023-05-31 by ivankravets