platformio/platformio-core-installer #2880: Installation Manager

nachannon, 2023-04-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.3
  PIO IDE: v3.1.1
  System: Linux, 5.19.0-41-generic, x64

  %23 Exception
  ```
  Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
 -> Installer version: 1.1.2
Platform: Linux-5.19.0-41-generic-x86_64-with-glibc2.35
Python version: 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/nachannon/.platformio/penv

	at /home/nachannon/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5303
	at ChildProcess.o (/home/nachannon/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at Socket.<anonymous> (node:internal/child_process:451:11)
	at Socket.emit (node:events:526:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

nachannon, 2023-05-10:
Yes had to select an eailer version of python then it installed

ivankravets, 2023-05-10:
Happy coding with PlatformIO!

closed 2023-05-10 by ivankravets