platformio/platformio-core-installer #2881: Installation Manager

HawazAbadi, 2023-04-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.3
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error:   ERROR: Error [WinError 225] Operation did not complete successfully because the file contains a virus or potentially unwanted software while executing command python setup.py egg_info
ERROR: Could not install packages due to an OSError: [WinError 225] Operation did not complete successfully because the file contains a virus or potentially unwanted software

Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\Hawaz Abadi\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\Hawaz Abadi\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Hawaz Abadi\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Downloading platformio-6.1.6.tar.gz (236 kB)
     -------------------------------------- 236.4/236.4 kB 2.4 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'

    at spawnOptions (c:\Users\Hawaz Abadi\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:226:25)
    at ChildProcess.o (c:\Users\Hawaz Abadi\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:170:13)
    at ChildProcess.emit (node:events:526:28)
    at ChildProcess.emit (node:domain:475:12)
    at maybeClose (node:internal/child_process:1092:16)
    at Socket.<anonymous> (node:internal/child_process:451:11)
    at Socket.emit (node:events:526:28)
    at Socket.emit (node:domain:475:12)
    at Pipe.<anonymous> (node:net:687:12)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/884

closed 2023-05-31 by ivankravets