platformio/platformio-core-installer #2883: Installation Manager FileNotFoundError

efficacy, 2023-04-28:
%23 Description of problem
The Installation manager fails to find a file while creating a virtual python environment.
Sadly, I can't find a clue as to _which_ file is not found, as it is not shown in the error trace :(

May be a duplicate of #2857, #2815, #2779, #2753, #2751 but there is no action on any of those either

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.3
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Frank\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "C:\Users\Frank\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "C:\Users\Frank\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\click\core.py", line 1637, in invoke
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\pioinstaller\core.py", line 70, in install_platformio_core
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\pioinstaller\core.py", line 98, in _install_platformio_core
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\pioinstaller\penv.py", line 71, in create_core_penv
  File "C:\Users\Frank\.platformio\.cache\tmp\.piocore-installer-n89oo_f3\tmpdiu6ubj0\pioinstaller.zip\pioinstaller\penv.py", line 141, in init_state
  File "C:\Users\Frank\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Frank\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Frank\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Frank\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\Frank\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Frank\.platformio\penv

	at c:\Users\Frank\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (c:\Users\Frank\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

efficacy, 2023-04-28:
Just in case anyone does read this. I eventually found a work-round.

It seems that the bug or conflict which is causing the problem is in the platformio core installation script, so neither the auto-install in the VS Code plugin nor installing the core using the script from https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html would work. Debugging that script was a PITA as it auto-deleted if anything went wrong, so I gave up.

To make Platformio IDE work on my system I had to do three things:
* install a recent Python + pip (in my case 3.11) and add both the main folder with python.exe and the Scripts folder to my path
* globally install platformio core using the "not recommended" pip method from https://docs.platformio.org/en/latest/core/installation/methods/pypi.html
* go into the settings for the PlatformIO plugin in VS Code and uncheck both **Use Builtin PIO Core** and **Use Builtin Python**

Only once I had done all three of those things did the PlatformIO IDE home page finally appear.

Phew!

ivankravets, 2023-05-10:
Do you use any suspicious antivirus tools?

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/884

closed 2023-05-31 by ivankravets