platformio/platformio-core-installer #2885: Installation Manager

Ulyssez, 2023-05-01:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.3
  PIO IDE: v3.1.1
  System: Linux, 4.9.253-tegra, arm64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Linux-4.9.253-tegra-aarch64-with-glibc2.25
Python version: 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/hakob/.platformio/penv
/usr/bin/python3

	at /home/hakob/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-arm64/node_modules/platformio-node-helpers/dist/index.js:1:5303
	at ChildProcess.o (/home/hakob/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-arm64/node_modules/platformio-node-helpers/dist/index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at Socket.<anonymous> (node:internal/child_process:451:11)
	at Socket.emit (node:events:526:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2955

closed 2023-05-31 by ivankravets