platformio/platformio-core-installer #2888: Installation Manager

jinyimi, 2023-05-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.3
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error:   ERROR: HTTP error 418 while getting https://pypi.doubanio.com/packages/cf/6f/6abff7fe6813e6f94129a50c8c70bf70fb33ed2515b1037e703cef6d7a7e/ajsonrpc-1.2.0-py3-none-any.whl%23sha256=0fa2c1cf8e619d18ffee96043822032d6520eda65d3b712f9540a3a63e9cac25 (from https://pypi.doubanio.com/simple/ajsonrpc/) (requires-python:>=3.5)
ERROR: Could not install requirement ajsonrpc==1.* from https://pypi.doubanio.com/packages/cf/6f/6abff7fe6813e6f94129a50c8c70bf70fb33ed2515b1037e703cef6d7a7e/ajsonrpc-1.2.0-py3-none-any.whl%23sha256=0fa2c1cf8e619d18ffee96043822032d6520eda65d3b712f9540a3a63e9cac25 (from platformio) because of HTTP error 418 Client Error: Unknown Status for url: https://pypi.doubanio.com/packages/cf/6f/6abff7fe6813e6f94129a50c8c70bf70fb33ed2515b1037e703cef6d7a7e/ajsonrpc-1.2.0-py3-none-any.whl for URL https://pypi.doubanio.com/packages/cf/6f/6abff7fe6813e6f94129a50c8c70bf70fb33ed2515b1037e703cef6d7a7e/ajsonrpc-1.2.0-py3-none-any.whl%23sha256=0fa2c1cf8e619d18ffee96043822032d6520eda65d3b712f9540a3a63e9cac25 (from https://pypi.doubanio.com/simple/ajsonrpc/) (requires-python:>=3.5)
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\16158\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\16158\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\16158\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Looking in indexes: https://pypi.doubanio.com/simple
Collecting platformio
  Using cached https://pypi.doubanio.com/packages/0b/49/0af5d39d7940ba1e6595f451b0d04712dddc4500db553af73b201d5edd4f/platformio-6.1.6.tar.gz (236 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting bottle==0.12.*
  Using cached https://pypi.doubanio.com/packages/bb/1f/5977ea88c6a3df6199db97d320e5da816d415d1eb75a987a1f6823d5cc9d/bottle-0.12.25-py3-none-any.whl (90 kB)
Collecting click<9,>=8.0.4
  Using cached https://pypi.doubanio.com/packages/c2/f1/df59e28c642d583f7dacffb1e0965d0e00b218e0186d7858ac5233dce840/click-8.1.3-py3-none-any.whl (96 kB)
Collecting colorama
  Using cached https://pypi.doubanio.com/packages/d1/d6/3965ed04c63042e047cb6a3e6ed1a63a35087b6a609aa3a15ed8ac56c221/colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting marshmallow==3.*
  Using cached https://pypi.doubanio.com/packages/ae/53/980a20d789029329fdf1546c315f9c92bf862c7f3e7294e3667afcc464f5/marshmallow-3.19.0-py3-none-any.whl (49 kB)
Collecting pyelftools<1,>=0.27
  Using cached https://pypi.doubanio.com/packages/04/7c/867630e6e6293793f838b31034aa1875e1c3bd8c1ec34a0929a2506f350c/pyelftools-0.29-py2.py3-none-any.whl (174 kB)
Collecting pyserial==3.5.*
  Using cached https://pypi.doubanio.com/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting requests==2.*
  Using cached https://pypi.doubanio.com/packages/d2/f4/274d1dbe96b41cf4e0efb70cbced278ffd61b5c7bb70338b62af94ccb25b/requests-2.28.2-py3-none-any.whl (62 kB)
Collecting semantic_version==2.10.*
  Using cached https://pypi.doubanio.com/packages/6a/23/8146aad7d88f4fcb3a6218f41a60f6c2d4e3a72de72da1825dc7c8f7877c/semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting tabulate==0.9.*
  Using cached https://pypi.doubanio.com/packages/40/44/4a5f08c96eb108af5cb50b41f76142f0afa346dfa99d5296fe7202a11854/tabulate-0.9.0-py3-none-any.whl (35 kB)
Collecting aiofiles==22.1.*
  Using cached https://pypi.doubanio.com/packages/a0/48/d5d1ab7cfe46e573c3694fa1365442a7d7cadc3abb03d8507e58a3755bb2/aiofiles-22.1.0-py3-none-any.whl (14 kB)
Collecting ajsonrpc==1.*

	at c:\Users\16158\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (c:\Users\16158\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

ivankravets, 2023-05-31:
> pypi.doubanio.com

Please avoid using espionage registries.

closed 2023-05-31 by ivankravets