platformio/platformio-core-installer #2895: Installation Manager

Runhan-Zhang, 2023-05-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.3
  PIO IDE: v3.1.1
  System: Linux, 5.15.0-69-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/runhan/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/home/runhan/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/home/runhan/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/pioinstaller/penv.py", line 58, in create_core_penv
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/pioinstaller/python.py", line 70, in fetch_portable_python
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/pioinstaller/python.py", line 97, in get_portable_python_url
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/requests/api.py", line 75, in get
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/requests/api.py", line 61, in request
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/requests/sessions.py", line 529, in request
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/requests/sessions.py", line 645, in send
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/requests/adapters.py", line 413, in send
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/requests/adapters.py", line 310, in get_connection
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/requests/adapters.py", line 183, in proxy_manager_for
  File "/home/runhan/.platformio/.cache/tmp/.piocore-installer-ftaau7g3/tmphx2v3s3y/pioinstaller.zip/requests/adapters.py", line 48, in SOCKSProxyManager
requests.exceptions.InvalidSchema: Missing dependencies for SOCKS support.
 -> Installer version: 1.1.2
Platform: Linux-5.15.0-69-generic-x86_64-with-glibc2.29
Python version: 3.8.10 (default, Mar 13 2023, 10:26:41) 
[GCC 9.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/runhan/.platformio/penv
/usr/bin/python3

    at spawnOptions (/home/runhan/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:226:25)
    at ChildProcess.o (/home/runhan/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:170:13)
    at ChildProcess.emit (node:events:526:28)
    at ChildProcess.emit (node:domain:475:12)
    at maybeClose (node:internal/child_process:1092:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

Runhan-Zhang, 2023-05-10:
这是来自QQ邮箱的假期自动回复邮件。你好，我已收到你的邮件 ：）

ivankravets, 2023-05-10:
Happy coding with PlatformIO!

closed 2023-05-10 by ivankravets