platformio/platformio-core-installer #2903: Installation Manager

AlexK799, 2023-05-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.78.0
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error:   ERROR: Error [WinError 225] Der Vorgang konnte nicht erfolgreich abgeschlossen werden, da die Datei einen Virus oder m�glicherweise unerw�nschte Software enth�lt while executing command python setup.py egg_info
ERROR: Could not install packages due to an OSError: [WinError 225] Der Vorgang konnte nicht erfolgreich abgeschlossen werden, da die Datei einen Virus oder m�glicherweise unerw�nschte Software enth�lt

Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\alexk\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\alexk\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\alexk\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-6.1.6.tar.gz (236 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'

	at c:\Users\alexk\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (c:\Users\alexk\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1093:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

AlexK799, 2023-05-10:
Hi, online Witze Otter computer i habe noch problems.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

ivankravets, 2023-05-10:
Happy coding with PlatformIO!

closed 2023-05-10 by ivankravets