platformio/platformio-core-installer #2905: Installation Manager

marcusobrien, 2023-05-07:
%23 Description of problem
I am doing a fresh install of VScode with PlatformIO. 

1. Unzip the vscode installer (as I dont want a single install, I want multiple installs - each with different plugins). This install is for embeded dev (hence the need for platforio)
2. I don't use a single Python install, I use diferent versions of Win Python, and these are not in my path. I start a WinPy shell to run python in the path if I need it, or I start a virtual environment shell depending what I want to use (run activate in the venv path etc)
3.  To run VSCode I cd to the folder that has the vscode installed executable (code.exe), and pass in as an argument, the folder that contains my workspace

cd U:\vscode\VSCode-win32-x64-1.69.0\EmbeddedC++
code.exe G:\EmbedWorkSpaces\Lora_Satellite

I make a data folder inside the vscode folder - so that vscode will install all plugins etc into that local data folder, otherwise it will put everything in /users/uname/.platformio - which I dont want, as this folder will be shared by all my vscode executables.

When I run the code.exe G:\EmbedWorkSpaces\Lora_Satellite, I expect it to install a local .platformio as there is a platformio.ini inside the project I am loading which comes from here

https://github.com/G4lile0/tinyGS

When I run vscode it complains with the error below - and I notice that there is a .platformio in my user folder (/users/uname/.platformio) - where that should be in the data folder AFAIK

` Directory of C:\Users\marcu\.platformio

2023-05-06  07:30 PM    <DIR>          .
2023-05-06  07:30 PM    <DIR>          ..
2023-05-06  07:21 PM    <DIR>          .cache
2023-05-06  07:30 PM    <DIR>          penv
2023-05-06  07:29 PM    <DIR>          python3`

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.69.0
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error:   WARNING: Building wheel for platformio failed: [WinError 3] The system cannot find the path specified: 'x:\\'
ERROR: Could not build wheels for platformio, which is required to install pyproject.toml-based projects
WARNING: There was an error checking the latest version of pip.
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\marcu\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\marcu\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\marcu\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Downloading platformio-6.1.6.tar.gz (236 kB)
     ------------------------------------- 236.4/236.4 kB 14.1 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting bottle==0.12.* (from platformio)
  Downloading bottle-0.12.25-py3-none-any.whl (90 kB)
     ---------------------------------------- 90.2/90.2 kB %3F eta 0:00:00
Collecting click<9,>=8.0.4 (from platformio)
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
     ---------------------------------------- 96.6/96.6 kB %3F eta 0:00:00
Collecting colorama (from platformio)
  Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting marshmallow==3.* (from platformio)
  Downloading marshmallow-3.19.0-py3-none-any.whl (49 kB)
     ---------------------------------------- 49.1/49.1 kB %3F eta 0:00:00
Collecting pyelftools<1,>=0.27 (from platformio)
  Downloading pyelftools-0.29-py2.py3-none-any.whl (174 kB)
     ---------------------------------------- 174.3/174.3 kB %3F eta 0:00:00
Collecting pyserial==3.5.* (from platformio)
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ---------------------------------------- 90.6/90.6 kB %3F eta 0:00:00
Collecting requests==2.* (from platformio)
  Downloading requests-2.30.0-py3-none-any.whl (62 kB)
     ---------------------------------------- 62.5/62.5 kB %3F eta 0:00:00
Collecting semantic_version==2.10.* (from platformio)
  Downloading semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting tabulate==0.9.* (from platformio)
  Downloading tabulate-0.9.0-py3-none-any.whl (35 kB)
Collecting aiofiles==22.1.* (from platformio)
  Downloading aiofiles-22.1.0-py3-none-any.whl (14 kB)
Collecting ajsonrpc==1.* (from platformio)
  Downloading ajsonrpc-1.2.0-py3-none-any.whl (22 kB)
Collecting starlette==0.23.* (from platformio)
  Downloading starlette-0.23.1-py3-none-any.whl (64 kB)
     ---------------------------------------- 64.5/64.5 kB %3F eta 0:00:00
Collecting uvicorn==0.20.* (from platformio)
  Downloading uvicorn-0.20.0-py3-none-any.whl (56 kB)
     ---------------------------------------- 56.9/56.9 kB %3F eta 0:00:00
Collecting wsproto==1.2.* (from platformio)
  Downloading wsproto-1.2.0-py3-none-any.whl (24 kB)
Collecting packaging>=17.0 (from marshmallow==3.*->platformio)
  Downloading packaging-23.1-py3-none-any.whl (48 kB)
     ---------------------------------------- 48.9/48.9 kB %3F eta 0:00:00
Collecting charset-normalizer<4,>=2 (from requests==2.*->platformio)
  Downloading charset_normalizer-3.1.0-cp39-cp39-win_amd64.whl (97 kB)
     ---------------------------------------- 97.1/97.1 kB 5.4 MB/s eta 0:00:00
Collecting idna<4,>=2.5 (from requests==2.*->platformio)
  Downloading idna-3.4-py3-none-any.whl (61 kB)
     ---------------------------------------- 61.5/61.5 kB %3F eta 0:00:00
Collecting urllib3<3,>=1.21.1 (from requests==2.*->platformio)
  Downloading urllib3-2.0.2-py3-none-any.whl (123 kB)
     ---------------------------------------- 123.2/123.2 kB %3F eta 0:00:00
Collecting certifi>=2017.4.17 (from requests==2.*->platformio)
  Downloading certifi-2022.12.7-py3-none-any.whl (155 kB)
     ---------------------------------------- 155.3/155.3 kB %3F eta 0:00:00
Collecting anyio<5,>=3.4.0 (from starlette==0.23.*->platformio)
  Downloading anyio-3.6.2-py3-none-any.whl (80 kB)
     ---------------------------------------- 80.6/80.6 kB %3F eta 0:00:00
Collecting typing-extensions>=3.10.0 (from starlette==0.23.*->platformio)
  Downloading typing_extensions-4.5.0-py3-none-any.whl (27 kB)
Collecting h11>=0.8 (from uvicorn==0.20.*->platformio)
  Downloading h11-0.14.0-py3-none-any.whl (58 kB)
     ---------------------------------------- 58.3/58.3 kB %3F eta 0:00:00
Collecting sniffio>=1.1 (from anyio<5,>=3.4.0->starlette==0.23.*->platformio)
  Downloading sniffio-1.3.0-py3-none-any.whl (10 kB)
Building wheels for collected packages: platformio
Failed to build platformio

	at u:\vscode\VSCode-win32-x64-1.69.0\EmbeddedC  \data\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (u:\vscode\VSCode-win32-x64-1.69.0\EmbeddedC  \data\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2023-05-10:
Could it be linked with https://github.com/platformio/platformio-core-installer/issues/2894 ?

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/884

closed 2023-05-31 by ivankravets