platformio/platformio-core-installer #2911: Installation Manager

siteswapjuggler, 2023-05-08:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.78.0
  PIO IDE: v3.1.1
  System: Darwin, 22.4.0, arm64

  %23 Exception
  ```
  Error: Error: Compatible PlatformIO Core not found.
Reason: Could not run `/Users/siteswapjuggler/.platformio/penv/bin/platformio --help`.
Error: Traceback (most recent call last):
  File "/Users/siteswapjuggler/.platformio/penv/bin/platformio", line 5, in <module>
    from platformio.__main__ import main
  File "/Users/siteswapjuggler/.platformio/penv/lib/python3.9/site-packages/platformio/__main__.py", line 21, in <module>
    from platformio import __version__, exception, maintenance
  File "/Users/siteswapjuggler/.platformio/penv/lib/python3.9/site-packages/platformio/maintenance.py", line 22, in <module>
    from platformio import __version__, app, exception, fs, telemetry
  File "/Users/siteswapjuggler/.platformio/penv/lib/python3.9/site-packages/platformio/telemetry.py", line 28, in <module>
    import requests
  File "/Users/siteswapjuggler/.platformio/penv/lib/python3.9/site-packages/requests/__init__.py", line 43, in <module>
    import urllib3
  File "/Users/siteswapjuggler/.platformio/penv/lib/python3.9/site-packages/urllib3/__init__.py", line 38, in <module>
    raise ImportError(
ImportError: urllib3 v2.0 only supports OpenSSL 1.1.1 , currently the 'ssl' module is compiled with LibreSSL 2.8.3. See: https://github.com/urllib3/urllib3/issues/2168

	at /Users/siteswapjuggler/.vscode/extensions/platformio.platformio-ide-3.1.1-darwin-arm64/node_modules/platformio-node-helpers/dist/index.js:1:5303
	at ChildProcess.o (/Users/siteswapjuggler/.vscode/extensions/platformio.platformio-ide-3.1.1-darwin-arm64/node_modules/platformio-node-helpers/dist/index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1093:16)
	at Socket.<anonymous> (node:internal/child_process:451:11)
	at Socket.emit (node:events:513:28)
	at Socket.emit (node:domain:489:12)
	at Pipe.<anonymous> (node:net:757:14)
  ```

ivankravets, 2023-05-10:
Fixed in the latest PlatformIO Core release. Please remove `/Users/siteswapjuggler/.platformio/penv` folder and restart VSCode.

closed 2023-05-10 by ivankravets

chrisbensen, 2023-05-11:
Correction, /Users/siteswapjuggler/.platformio/penv, it should be /Users/$USER/.platformio/penv

siteswapjuggler, 2023-05-12:
Actually I tried that and it doesn't work, anyway on the last install it installed correctly. I suspect that another program could conflict with the ssl ressource involved in this error.