platformio/platformio-core-installer #2914: Installation Manager

duksubbc, 2023-05-09:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.77.3
  PIO IDE: v3.1.1
  System: Linux, 5.10.102.1-microsoft-standard-WSL2, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Linux-5.10.102.1-microsoft-standard-WSL2-x86_64-with-glibc2.25
Python version: 3.6.9 (default, Mar 15 2022, 13:55:28) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/ds/.platformio/penv
/usr/bin/python3

	at /home/ds/.vscode-server/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5241
	at ChildProcess.o (/home/ds/.vscode-server/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4539)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

ivankravets, 2023-05-10:
Have you resolved this issue?

ivankravets, 2023-05-31:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2955

closed 2023-05-31 by ivankravets