platformio/platformio-core-installer #2921: Installation Manager

jpparmar14, 2023-05-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.74.2
  PIO IDE: v2.99.4
  System: Linux, 5.15.0-58-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/urllib3/connectionpool.py", line 700, in urlopen
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/urllib3/connectionpool.py", line 994, in _prepare_proxy
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/urllib3/connection.py", line 369, in connect
  File "/usr/lib/python3.8/http/client.py", line 905, in _tunnel
    raise OSError("Tunnel connection failed: %d %s" % (code,
OSError: Tunnel connection failed: 403 Forbidden

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/requests/adapters.py", line 440, in send
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/urllib3/connectionpool.py", line 785, in urlopen
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/urllib3/util/retry.py", line 592, in increment
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/python-portable (Caused by ProxyError('Cannot connect to proxy.', OSError('Tunnel connection failed: 403 Forbidden')))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aatef/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/home/aatef/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/home/aatef/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/pioinstaller/penv.py", line 58, in create_core_penv
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/pioinstaller/python.py", line 70, in fetch_portable_python
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/pioinstaller/python.py", line 97, in get_portable_python_url
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/requests/api.py", line 75, in get
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/requests/api.py", line 61, in request
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/requests/sessions.py", line 529, in request
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/requests/sessions.py", line 645, in send
  File "/home/aatef/.platformio/.cache/tmp/.piocore-installer-9apd99c4/tmpoc3mdx6p/pioinstaller.zip/requests/adapters.py", line 513, in send
requests.exceptions.ProxyError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/python-portable (Caused by ProxyError('Cannot connect to proxy.', OSError('Tunnel connection failed: 403 Forbidden')))
 -> Installer version: 1.1.2
Platform: Linux-5.15.0-58-generic-x86_64-with-glibc2.29
Python version: 3.8.10 (default, Nov 14 2022, 12:59:47) 
[GCC 9.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/aatef/.platformio/penv
/usr/bin/python3

	at /home/aatef/.vscode/extensions/platformio.platformio-ide-2.99.4/node_modules/platformio-node-helpers/dist/index.js:1:5362
	at ChildProcess.o (/home/aatef/.vscode/extensions/platformio.platformio-ide-2.99.4/node_modules/platformio-node-helpers/dist/index.js:1:4660)
	at ChildProcess.emit (node:events:526:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1092:16)
	at Socket.<anonymous> (node:internal/child_process:451:11)
	at Socket.emit (node:events:526:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

ivankravets, 2023-05-31:
Do you use a proxy?

ivankravets, 2023-06-02:
Please re-test with the pre-release version. Navigate to the PlatformIO IDE extension in the VSCode and "Switch to the Pre-Release". Does it work now?
[image]

closed 2023-06-02 by ivankravets