platformio/platformio-core-installer #2947: installs but after restart not installed

ktmer02, 2023-05-24:
Hey guys, so im trying to install PlatformIO.
The first issue is that when i try to install it it says "python not installed" but it is installed so i put in the path to python.exe.
Then it installs and says restart Visual Studios Code, after doing that it trys to install again and again says "python not installed"
Can someone please help me out with this.

Best,
David

ivankravets, 2023-05-31:
What is your OS?

ktmer02, 2023-05-31:
> 

Windows 10

ivankravets, 2023-05-31:
Do you use any espionage software("antivirus")?

ivankravets, 2023-06-02:
Please re-test with the pre-release version. Navigate to the PlatformIO IDE extension in the VSCode and "Switch to the Pre-Release". Does it work now?
[image]

closed 2023-06-02 by ivankravets