platformio/platformio-core-installer #2948: Installation Manager

Kili816, 2023-05-24:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.78.2
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error: ERROR: Could not install packages due to an OSError: [Errno 2] No such file or directory: 'D:\\ROPB\\Vs-Code\\VSCode-win32-x64-1.78.2\\data\\extensions\\platformio\\.cache\\tmp\\.piocore-installer-jy86jeed\\pip-install-4arb2853\\platformio_e592fd2fa7fe4cd39682a58c9da9a233\\platformio/assets/templates/ide-projects/eclipse/.settings/PlatformIO Debugger.launch.tpl'

Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['D:\\ROPB\\Vs-Code\\VSCode-win32-x64-1.78.2\\data\\extensions\\platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: D:\ROPB\Vs-Code\VSCode-win32-x64-1.78.2\data\extensions\platformio\python3\python.exe
Creating a virtual environment at D:\ROPB\Vs-Code\VSCode-win32-x64-1.78.2\data\extensions\platformio\penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-6.1.7.tar.gz (239 kB)

	at d:\ROPB\Vs-Code\VSCode-win32-x64-1.78.2\data\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (d:\ROPB\Vs-Code\VSCode-win32-x64-1.78.2\data\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1112:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

ivankravets, 2023-05-31:
Do you use any espionage software such as "antivirus"?

ivankravets, 2023-06-02:
Please re-test with the pre-release version. Navigate to the PlatformIO IDE extension in the VSCode and "Switch to the Pre-Release". Does it work now?
[image]