platformio/platformio-core-installer #2955: Installation Manager

Reza-Safari-dev, 2023-05-25:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.78.2
  PIO IDE: v3.1.1
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\Reza Safari\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Reza Safari\.platformio\penv

	at c:\Users\Reza Safari\.vscode\extensions\platformio.platformio-ide-3.1.1\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (c:\Users\Reza Safari\.vscode\extensions\platformio.platformio-ide-3.1.1\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1112:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

garethbrown, 2023-06-05:
I've just had this issue on Windows. Fixed it by upgrading to Python version: 3.10.11.

amdialloge1, 2023-06-13:
Here is the problem i am facing and i have used commands provide in the installer script but still i have the same problem : 
Error: Error: The Python 3.6.9 (/usr/bin/python3) interpreter is not compatible.
Reason: Could not find distutils module

	at /home/adiallo/.vscode/extensions/platformio.platformio-ide-3.0.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5362
	at ChildProcess.o (/home/adiallo/.vscode/extensions/platformio.platformio-ide-3.0.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4660)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at Socket.<anonymous> (node:internal/child_process:479:11)
	at Socket.emit (node:events:513:28)
	at Socket.emit (node:domain:489:12)
	at Pipe.<anon...
 I wanted to know if there is any other solution i can use. thank you!

iottrends, 2023-07-05:
Same issue , platformio core install fails within visual studio... 
RAN
**python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"**
Installer version: 1.1.3
Platform: Linux-5.0.0-23-generic-x86_64-with-glibc2.25
Python version: 3.6.8 (default, Jan 14 2019, 11:02:34) 
[GCC 8.0.1 20180414 (experimental) [trunk revision 259383]]
Python path: /usr/bin/python3
Creating a virtual environment at /home/labuser/.platformio/penv
/usr/bin/python3
Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
Traceback (most recent call last):
  File "<string>", line 105, in <module>
  File "<string>", line 101, in main
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '/tmp/tmpbanimt4n']' returned non-zero exit status 1.
labuser@labuser-OptiPlex-7040:~/Downloads$

AlphaB404, 2023-07-06:
I've got same issue. install Plaformio Core on jetson nano.

python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
Installer version: 1.1.3
Platform: Linux-4.9.253-tegra-aarch64-with-glibc2.25
Python version: 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/wcrp_tgi/.platformio/penv
/usr/bin/python3
Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
Traceback (most recent call last):
  File "<string>", line 105, in <module>
  File "<string>", line 101, in main
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '/tmp/tmp18hd2ohi']' returned non-zero exit status 1.

Richsl, 2023-08-18:
I'm getting the same error both in VSCode and when I try to install from terminal...

PS C:\Users\richa\Documents\PlatformIO> python.exe get-platformio.py
Installer version: 1.2.1
Platform: Windows-10
Python version: 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)]
Python path: C:\Users\richa\.platformio\penv\Scripts\python.exe
Creating a virtual environment at C:\Users\richa\.platformio\penv
C:\Users\richa\.platformio\penv\Scripts\python.exe
Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
PS C:\Users\richa\Documents\PlatformIO>

vahidajalluian, 2023-09-28:
I get this one
Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.2.1
Platform: Windows-10
Python version: 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)]
Python path: C:\Users\vahid\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\vahid\.platformio\penv
C:\Users\vahid\.platformio\python3\python.exe

	at c:\Users\vahid\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5300
	at ChildProcess.c (c:\Users\vahid\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64...