platformio/platformio-core-installer #2960: Installation Manager

santisaran, 2023-05-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.78.2
  PIO IDE: v3.1.1
  System: Linux, 5.19.0-42-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/saran/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/home/saran/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/home/saran/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/saran/.platformio/.cache/tmp/.piocore-installer-or028zdq/tmpy46qqa7o/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/saran/.platformio/penv/bin/python'
 -> Installer version: 1.1.2
Platform: Linux-5.19.0-42-generic-x86_64-with-glibc2.35
Python version: 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/saran/.platformio/penv

	at /home/saran/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5303
	at ChildProcess.o (/home/saran/.vscode/extensions/platformio.platformio-ide-3.1.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1112:16)
	at Socket.<anonymous> (node:internal/child_process:470:11)
	at Socket.emit (node:events:513:28)
	at Socket.emit (node:domain:489:12)
	at Pipe.<anonymous> (node:net:757:14)
  ```

ivankravets, 2023-05-31:
What is your Linux distributive version?

santisaran, 2023-05-31:
`Linux santi 5.19.0-42-generic #43~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Apr 21 16:51:08 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
`
If I execute get-platformio-1.1.2.py on terminal, with ~/platformio/penv python, run ok:
```
 ~/.platformio/.cache/tmp  source ../../penv/local/bin/activate                                                                                                                                    1 ✘  09:36:29
 ~/.platformio/.cache/tmp  python get-platformio-1.1.2.py                                                                                                                                  ✔  penv Py  09:36:49
Installer version: 1.1.2
Platform: Linux-5.19.0-42-generic-x86_64-with-glibc2.35
Python version: 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
Python path: /home/saran/.platformio/penv/local/bin/python
Creating a virtual environment at /home/saran/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-6.1.7-py3-none-any.whl
Collecting bottle==0.12.* (from platformio)
  Using cached bottle-0.12.25-py3-none-any.whl (90 kB)
Collecting click<9,>=8.0.4 (from platformio)
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting colorama (from platformio)
  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting marshmallow==3.* (from platformio)
  Using cached marshmallow-3.19.0-py3-none-any.whl (49 kB)
Collecting pyelftools<1,>=0.27 (from platformio)
  Using cached pyelftools-0.29-py2.py3-none-any.whl (174 kB)
Collecting pyserial==3.5.* (from platformio)
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting requests==2.* (from platformio)
  Using cached requests-2.31.0-py3-none-any.whl (62 kB)
Collecting urllib3<2 (from platformio)
  Using cached urllib3-1.26.16-py2.py3-none-any.whl (143 kB)
Collecting semantic-version==2.10.* (from platformio)
  Using cached semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting tabulate==0.9.* (from platformio)
  Using cached tabulate-0.9.0-py3-none-any.whl (35 kB)
Collecting aiofiles==23.1.* (from platformio)
  Using cached aiofiles-23.1.0-py3-none-any.whl (14 kB)
Collecting ajsonrpc==1.* (from platformio)
  Using cached ajsonrpc-1.2.0-py3-none-any.whl (22 kB)
Collecting starlette==0.26.* (from platformio)
  Using cached starlette-0.26.1-py3-none-any.whl (66 kB)
Collecting uvicorn==0.22.* (from platformio)
  Using cached uvicorn-0.22.0-py3-none-any.whl (58 kB)
Collecting wsproto==1.2.* (from platformio)
  Using cached wsproto-1.2.0-py3-none-any.whl (24 kB)
Collecting packaging>=17.0 (from marshmallow==3.*->platformio)
  Using cached packaging-23.1-py3-none-any.whl (48 kB)
Collecting charset-normalizer<4,>=2 (from requests==2.*->platformio)
  Using cached charset_normalizer-3.1.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (197 kB)
Collecting idna<4,>=2.5 (from requests==2.*->platformio)
  Using cached idna-3.4-py3-none-any.whl (61 kB)
Collecting certifi>=2017.4.17 (from requests==2.*->platformio)
  Using cached certifi-2023.5.7-py3-none-any.whl (156 kB)
Collecting anyio<5,>=3.4.0 (from starlette==0.26.*->platformio)
  Using cached anyio-3.7.0-py3-none-any.whl (80 kB)
Collecting h11>=0.8 (from uvicorn==0.22.*->platformio)
  Using cached h11-0.14.0-py3-none-any.whl (58 kB)
Collecting sniffio>=1.1 (from anyio<5,>=3.4.0->starlette==0.26.*->platformio)
  Using cached sniffio-1.3.0-py3-none-any.whl (10 kB)
Installing collected packages: pyserial, pyelftools, bottle, urllib3, tabulate, sniffio, semantic-version, packaging, idna, h11, colorama, click, charset-normalizer, certifi, ajsonrpc, aiofiles, wsproto, uvicorn, requests, marshmallow, anyio, starlette, platformio
Successfully installed aiofiles-23.1.0 ajsonrpc-1.2.0 anyio-3.7.0 bottle-0.12.25 certifi-2023.5.7 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 h11-0.14.0 idna-3.4 marshmallow-3.19.0 packaging-23.1 platformio-6.1.7 pyelftools-0.29 pyserial-3.5 requests-2.31.0 semantic-version-2.10.0 sniffio-1.3.0 starlette-0.26.1 tabulate-0.9.0 urllib3-1.26.16 uvicorn-0.22.0 wsproto-1.2.0

PlatformIO Core has been successfully installed into an isolated environment `/home/saran/.platformio/penv`!

The full path to `platformio.exe` is `/home/saran/.platformio/penv/bin/platformio`

If you need an access to `platformio.exe` from other applications, please install Shell Commands
(add PlatformIO Core binary directory `/home/saran/.platformio/penv/bin` to the system environment PATH variable):

See https://docs.platformio.org/page/installation.html#install-shell-commands

```

santisaran, 2023-05-31:
Ye, I remove .platformio, but visual code create that again.
That is visual code output:
```
Installing PlatformIO IDE...
It may take a few minutes depending on your connection speed
Please do not close this window and do not open other folders until this process is completed.

Debugging information is available via VSCode > Help > Toggle Developer Tools > Console.
Failed to install PlatformIO IDE.
```
And the error traceback is the same

santisaran, 2023-05-31:
Yes, because the script search for python in penv/bin but is in  .platformio/penv/local/bin/python   
` FileNotFoundError: [Errno 2] No such file or directory: '/home/saran/.platformio/penv/bin/python'  `
```
rm -R  ~/.platformio                                                                                                                                                                             ~  python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"                                                                             
Installer version: 1.1.3
Platform: Linux-5.19.0-42-generic-x86_64-with-glibc2.35
Python version: 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/saran/.platformio/penv
Traceback (most recent call last):
  File "/tmp/tmph0oyoaiq", line 68, in <module>
    main()
  File "/tmp/tmph0oyoaiq", line 60, in main
    bootstrap()
  File "/tmp/tmph0oyoaiq", line 46, in bootstrap
    pioinstaller.__main__.main()
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/click/core.py", line 1053, in main
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/tmp/.piocore-installer-evr1ed8s/tmpg_0rbaqr/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/saran/.platformio/penv/bin/python'
Traceback (most recent call last):
  File "<string>", line 105, in <module>
  File "<string>", line 101, in main
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '/tmp/tmph0oyoaiq']' returned non-zero exit status 1.
```

santisaran, 2023-05-31:
I retry with https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py and works fine.
Yes, apparently I have problems with virtualenv.

Thanks.

closed 2023-05-31 by ivankravets