platformio/platformio-core-installer #2968: Failed Install When Opening Project in VSCode

a-thansen, 2023-06-02:
Was trying to open a project in VS Code w./ platformIO, expecting it to install.

However, I receive the following error messages:

```
Error:   error: subprocess-exited-with-error
  
  python setup.py bdist_wheel did not run successfully.
  exit code: 1
  
  [425 lines of output]
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib
  creating build\lib\platformio
  copying platformio\app.py -> build\lib\platformio
  copying platformio\cache.py -> build\lib\platformio
  copying platformio\cli.py -> build\lib\platformio
  copying platformio\compat.py -> build\lib\platformio
  copying platformio\exception.py -> build\lib\platformio
  copying platformio\fs.py -> build\lib\platformio
  copying platformio\http.py -> build\lib\platformio
  copying platformio\maintenance....
```
```
Error: Could not start PIO Home server: Timeout error
	at Timeout._onTimeout (c:\Users\Anastasia T Hansen\.vscode\extensions\platformio.platformio-ide-3.1.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10444)
	at listOnTimeout (node:internal/timers:559:17)
	at process.processTimers (node:internal/timers:502:7)...
```
This is what the output terminal says:

```
Installing PlatformIO IDE...
It may take a few minutes depending on your connection speed
Please do not close this window and do not open other folders until this process is completed.

Debugging information is available via VSCode > Help > Toggle Developer Tools > Console.
Failed to install PlatformIO IDE.
```

My system specs are:
- Windows 10

The visual studio code specs: 
```
Version: 1.78.2 (user setup)
Commit: b3e4e68a0bc097f0ae7907b217c1119af9e03435
Date: 2023-05-10T14:39:26.248Z
Electron: 22.5.2
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.19044
Sandboxed: Yes
```
PlatformIO v3.1.1

Please help, thank you!!! 
I've already tried to enable long windows paths, does not seem to work though.

a-thansen, 2023-06-02:
[message.txt](https://github.com/platformio/platformio-core-installer/files/11635345/message.txt)
 Full error message.

FuzzyLitchi, 2023-06-02:
I'd like to add that following the instructions from #2894 did not resolve the issue, just to make it clear. (I was assisting in getting platformIO to work on her machine)

ivankravets, 2023-06-06:
Please re-test with the pre-release version. Navigate to the PlatformIO IDE extension in the VSCode and "Switch to the Pre-Release". Does it work now?
[image]

closed 2023-06-06 by ivankravets

FuzzyLitchi, 2023-06-08:
The issue was already resolved, by installing python using the windows installer. The previous message by a-thasen has a typo, it's supposed to say "Installing Python 3 and changing the path *length* using the windows installer resolved the issue." This behavior is documented [here](https://docs.python.org/3/using/windows.html#removing-the-max-path-limitation)

It's not clear to me exactly what the Python installation executable does, or which change fixed the error. Therefore, it is not trivial to return the system to a state where we get the same error, so we cannot test if the pre-release would fix the issue. If you would like to test it yourself, I think a stock windows 10 installation should be sufficient to reproduce the issue.