platformio/platformio-core-installer #2975: Installation Manager

teeceeel, 2023-06-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.78.2
  PIO IDE: v3.1.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error:   ERROR: Error [WinError 225] Operation did not complete successfully because the file contains a virus or potentially unwanted software while executing command python setup.py egg_info
ERROR: Could not install packages due to an OSError: [WinError 225] Operation did not complete successfully because the file contains a virus or potentially unwanted software

Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\TC\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\TC\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\TC\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-6.1.7.tar.gz (239 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'

	at c:\Users\TC\.vscode\extensions\platformio.platformio-ide-3.1.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5303
	at ChildProcess.o (c:\Users\TC\.vscode\extensions\platformio.platformio-ide-3.1.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4601)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1112:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

ivankravets, 2023-06-06:
Please re-test with the pre-release version. Navigate to the PlatformIO IDE extension in the VSCode and "Switch to the Pre-Release". Does it work now?
[image]

closed 2023-06-06 by ivankravets

teeceeel, 2023-06-06:
It is already resolved, I found out the culprit were softwares from Reasonlabs (which got installed when Windows 10 updates itself), I uninstalled 3 softwares by Reasonlabs, the VPN, Anti virus and blahblah check endpoint (I forgot exactly what it said)
Another thing: after PlatformIO successfully installed I need to exclude it from Norton Anti Virus checks in order for it to perform the Run commands

Hopefully this will help other who got the same problem when installing in Windows 10/

TC

teeceeel, 2023-06-06:
Hi thank you for your email,

It is already resolved, I found out the culprit were softwares from
Reasonlabs (which got installed when Windows 10 updates itself), I
uninstalled 3 softwares by Reasonlabs, the VPN, Anti virus and blahblah
check endpoint (I forgot exactly what it said)

Another thing: after PlatformIO successfully installed I need to exclude it
from Norton Anti Virus checks in order for it to perform the Run commands

Besh Wishes,
TC