platformio/platformio-core-installer #2986: Installation Manager

KENCUPO, 2023-06-10:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.79.0
  PIO IDE: v3.2.0
  System: Linux, 5.13.0-40-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/__main__.py", line 29, in <module>
    from pip._internal.cli.main import main as _main
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/cli/autocompletion.py", line 10, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/cli/main_parser.py", line 9, in <module>
    from pip._internal.build_env import get_runnable_pip
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/build_env.py", line 19, in <module>
    from pip._internal.cli.spinners import open_spinner
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/cli/spinners.py", line 9, in <module>
    from pip._internal.utils.logging import get_indentation
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/utils/logging.py", line 29, in <module>
    from pip._internal.utils.misc import ensure_dir
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/utils/misc.py", line 44, in <module>
    from pip._internal.locations import get_major_minor_version
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/locations/__init__.py", line 66, in <module>
    from . import _distutils
  File "/home/ken/.platformio/penv/lib/python3.9/site-packages/pip/_internal/locations/_distutils.py", line 20, in <module>
    from distutils.cmd import Command as DistutilsCommand
ModuleNotFoundError: No module named 'distutils.cmd'
Error: Could not install PlatformIO Core: Command '['/home/ken/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.3
Platform: Linux-5.13.0-40-generic-x86_64-with-glibc2.34
Python version: 3.9.7 (default, Sep 10 2021, 14:59:43) 
[GCC 11.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/ken/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core

    at /home/ken/.vscode/extensions/platformio.platformio-ide-3.2.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5300
    at ChildProcess.o (/home/ken/.vscode/extensions/platformio.platformio-ide-3.2.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4598)
    at ChildProcess.emit (node:events:513:28)
    at ChildProcess.emit (node:domain:489:12)
    at maybeClose (node:internal/child_process:1121:16)
    at Socket.<anonymous> (node:internal/child_process:479:11)
    at Socket.emit (node:events:513:28)
    at Socket.emit (node:domain:489:12)
    at Pipe.<anonymous> (node:net:757:14)
  ```

ivankravets, 2023-06-22:
Does `apt install python3-distutils` fix the issue?

ivankravets, 2023-06-22:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/85

closed 2023-06-22 by ivankravets