platformio/platformio-core-installer #3003: Installation Manager

jfaginas, 2023-06-23:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.79.2
  PIO IDE: v3.2.0
  System: Linux, 5.15.0-75-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/jfaginas/.platformio/.cache/tmp/get-platformio-1.1.3.py", line 69, in <module>
    main()
  File "/home/jfaginas/.platformio/.cache/tmp/get-platformio-1.1.3.py", line 61, in main
    bootstrap()
  File "/home/jfaginas/.platformio/.cache/tmp/get-platformio-1.1.3.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/jfaginas/.platformio/.cache/tmp/.piocore-installer-m_jeseua/tmpcr4vwyz7/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/jfaginas/.platformio/penv/bin/python'
 -> Installer version: 1.1.3
Platform: Linux-5.15.0-75-generic-x86_64-with-glibc2.35
Python version: 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/jfaginas/.platformio/penv

	at /home/jfaginas/.vscode/extensions/platformio.platformio-ide-3.2.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5300
	at ChildProcess.o (/home/jfaginas/.vscode/extensions/platformio.platformio-ide-3.2.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at Socket.<anonymous> (node:internal/child_process:479:11)
	at Socket.emit (node:events:513:28)
	at Socket.emit (node:domain:489:12)
	at Pipe.<anonymous> (node:net:757:14)
  ```

jfaginas, 2023-06-23:
I tried removing /home/jfaginas/.platformio/penv folder and restart VSCode without success.Thx in advance!

ivankravets, 2023-07-05:
Please provide an output of 
```
ls -al /home/jfaginas/.platformio/penv/bin/
```

SamuelHill, 2023-07-05:
I am having the same issue on Windows and there is no `C:\Users\Sam Hill\.platformio\penv\bin`

SamuelHill, 2023-07-05:
The manual [installer script](https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html) worked for me! [Referenced here](https://github.com/platformio/platformio-core-installer/issues/2955#issue-1726313038)

One note - the `python.exe get-platformio.py` needs to be `C:\path\to\.platformio\python3\python.exe get-platformio.py` otherwise you will be using the default python on your system.

ivankravets, 2023-07-07:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/85

Please `sudo apt-get install python3-venv python3-distutils` and restart VSCode.

closed 2023-07-07 by ivankravets