platformio/platformio-core-installer #3010: Installation Manager

WPluim, 2023-07-01:
Description of problem
continuation from #2991
Your answer was: 'It seems you use the espionage PyPi index. Please use official https://pypi.org/'.
My problem is now: Sorry, I don't understand what you mean to tell me. I think I have no choice in the process of installing / updating.
I am using MS VSCode (1.79.2) on MS Windows 10. VSCode wants to install or update the PlatformIO IDE.
The PlatformIO Installer starts with downloading a portable Python interpreter for PlatformIO.
An after that it fails in installing the PlatformIO Core. 
OUTPUT says: Debugging information is available via VSCode > Help > Toggle Developer Tools > Console. 
                        Failed to install PlatformIO IDE.

Console says:
-------------------------------
TMScopeRegistry.ts:45 
Overwriting grammar scope name to file mapping for scope source.cpp.embedded.macro.
Old grammar file: file:///c%3A/Users/Pluim/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/extensions/cpp/syntaxes/cpp.embedded.macro.tmLanguage.json.
New grammar file: file:///c%3A/Users/Pluim/.vscode/extensions/jeff-hykin.better-cpp-syntax-1.17.2/autogenerated/cpp.embedded.macro.tmLanguage.json
register @ TMScopeRegistry.ts:45
TMScopeRegistry.ts:45 
Overwriting grammar scope name to file mapping for scope source.cpp.
Old grammar file: file:///c%3A/Users/Pluim/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/extensions/cpp/syntaxes/cpp.tmLanguage.json.
New grammar file: file:///c%3A/Users/Pluim/.vscode/extensions/jeff-hykin.better-cpp-syntax-1.17.2/autogenerated/cpp.tmLanguage.json
register @ TMScopeRegistry.ts:45
console.ts:137 [Extension Host] activating extension
console.ts:137 [Extension Host] starting language server
console.ts:137 [Extension Host] PlatformIO IDE Global State Object
log.ts:441   ERR [Extension Host] (node:8624) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
(Use `Code --trace-deprecation ...` to show where the warning was created)
console.ts:137 [Extension Host] (node:8624) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
(Use `Code --trace-deprecation ...` to show where the warning was created)
y @ console.ts:137
console.ts:137 [Extension Host] runCommand C:\Users\Pluim\.platformio\python3\python.exe Array(3) Object
log.ts:421  INFO [perf] Render performance baseline is 118ms
console.ts:137 [Extension Host] C:\Users\Pluim\.platformio\python3\python.exe Error: 
	at c:\Users\Pluim\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5362
	at ChildProcess.o (c:\Users\Pluim\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4660)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at Socket.<anonymous> (node:internal/child_process:479:11)
	at Socket.emit (node:events:513:28)
	at Socket.emit (node:domain:489:12)
	at Pipe.<anonymous> (node:net:757:14)
y @ console.ts:137
console.ts:137 [Extension Host] runCommand C:\Python311\python.exe Array(3) Object
console.ts:137 [Extension Host] runCommand C:\Python311\python.exe Array(8) Object
console.ts:137 [Extension Host] Error: Error: Compatible PlatformIO Core not found.
Reason: PlatformIO executable not found in `C:\Users\Pluim\.platformio\penv\Scripts`

	at c:\Users\Pluim\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5362
	at ChildProcess.o (c:\Users\Pluim\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4660)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
y @ console.ts:137
log.ts:441   ERR Error: ENOENT: no such file or directory, stat 'c:\Users\Pluim\AppData\Roaming\Code\logs\20230701T155026\window1\exthost\output_logging_20230701T155029\4-PlatformIO Installation.log': EntryNotFound (FileSystemError): Error: ENOENT: no such file or directory, stat 'c:\Users\Pluim\AppData\Roaming\Code\logs\20230701T155026\window1\exthost\output_logging_20230701T155029\4-PlatformIO Installation.log'
    at n.create (C:\Users\Pluim\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\code\electron-main\main.js:37:4334)
    at i (C:\Users\Pluim\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\code\electron-main\main.js:37:4464)
    at t.bb (C:\Users\Pluim\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\code\electron-main\main.js:42:40412)
    at t.stat (C:\Users\Pluim\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\code\electron-main\main.js:42:32627)
log.ts:441   ERR Unable to read file 'c:\Users\Pluim\AppData\Roaming\Code\logs\20230701T155026\window1\exthost\output_logging_20230701T155029\4-PlatformIO Installation.log' (Error: Unable to resolve nonexistent file 'c:\Users\Pluim\AppData\Roaming\Code\logs\20230701T155026\window1\exthost\output_logging_20230701T155029\4-PlatformIO Installation.log'): Error: Unable to read file 'c:\Users\Pluim\AppData\Roaming\Code\logs\20230701T155026\window1\exthost\output_logging_20230701T155029\4-PlatformIO Installation.log' (Error: Unable to resolve nonexistent file 'c:\Users\Pluim\AppData\Roaming\Code\logs\20230701T155026\window1\exthost\output_logging_20230701T155029\4-PlatformIO Installation.log')
    at e.FileService.w (vscode-file://vscode-app/c:/Users/Pluim/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:624:18204)
    at e.FileService.v (vscode-file://vscode-app/c:/Users/Pluim/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:624:17865)
    at async e.FileService.u (vscode-file://vscode-app/c:/Users/Pluim/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:624:17147)
    at async e.FileOutputChannelModel.I (vscode-file://vscode-app/c:/Users/Pluim/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:1638:39935)
    at async vscode-file://vscode-app/c:/Users/Pluim/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:1638:39131
console.ts:137 [Extension Host] Using predownloaded package from c:\Users\Pluim\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\assets\predownloaded\python-portable-windows_amd64-1.30910.0.tar.gz
log.ts:441   ERR [Extension Host] (node:8624) [DEP0147] DeprecationWarning: In future versions of Node.js, fs.rmdir(path, { recursive: true }) will be removed. Use fs.rm(path, { recursive: true }) instead
console.ts:137 [Extension Host] (node:8624) [DEP0147] DeprecationWarning: In future versions of Node.js, fs.rmdir(path, { recursive: true }) will be removed. Use fs.rm(path, { recursive: true }) instead
y @ console.ts:137
[... 50 lines cut ...]
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\Pluim\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\Pluim\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Pluim\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached https://files.pythonhosted.org/packages/d7/64/e606dc0a4aa68352bc2147a3f1a1fdb517af1063dcd9b908603036020396/platformio-6.1.7-py3-none-any.whl
Collecting marshmallow==3.* (from platformio)
  Using cached https://files.pythonhosted.org/packages/ec/f3/62015688fdb1e02779702c36e500a123e019a4f99a70fd849de2c7572748/marshmallow-3.14.1-py3-none-any.whl
Collecting colorama (from platformio)
[... cut ...]
	at c:\Users\Pluim\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5362
	at ChildProcess.o (c:\Users\Pluim\.vscode\extensions\platformio.platformio-ide-3.0.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4660)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

WPluim, 2023-07-01:
In addition:
Sorry, I don't understand what you mean by 'espionage'? It must be the installer that makes the choice?
I'm using MS VS Code with PlatformIO. PlatformIO wants to update itself. That does not function properly.
I can update on the CLI/Powershell and now all requirements are satisfied, but it still does not work.

PS C:\Users\Pluim.platformio\penv\Scripts> pip install platformio
Requirement already satisfied: platformio in c:\python311\lib\site-packages (6.1.7)
Requirement already satisfied: bottle==0.12.* in c:\python311\lib\site-packages (from platformio) (0.12.25)
Requirement already satisfied: click<9,>=8.0.4 in c:\python311\lib\site-packages (from platformio) (8.1.3)
Requirement already satisfied: colorama in c:\python311\lib\site-packages (from platformio) (0.4.6)
Requirement already satisfied: marshmallow==3.* in c:\python311\lib\site-packages (from platformio) (3.19.0)
Requirement already satisfied: pyelftools<1,>=0.27 in c:\python311\lib\site-packages (from platformio) (0.29)
Requirement already satisfied: pyserial==3.5.* in c:\python311\lib\site-packages (from platformio) (3.5)
Requirement already satisfied: requests==2.* in c:\python311\lib\site-packages (from platformio) (2.31.0)
Requirement already satisfied: urllib3<2 in c:\python311\lib\site-packages (from platformio) (1.26.16)
Requirement already satisfied: semantic-version==2.10.* in c:\python311\lib\site-packages (from platformio) (2.10.0)
Requirement already satisfied: tabulate==0.9.* in c:\python311\lib\site-packages (from platformio) (0.9.0)
Requirement already satisfied: aiofiles==23.1.* in c:\python311\lib\site-packages (from platformio) (23.1.0)
Requirement already satisfied: ajsonrpc==1.* in c:\python311\lib\site-packages (from platformio) (1.2.0)
Requirement already satisfied: starlette==0.26.* in c:\python311\lib\site-packages (from platformio) (0.26.1)
Requirement already satisfied: uvicorn==0.22.* in c:\python311\lib\site-packages (from platformio) (0.22.0)
Requirement already satisfied: wsproto==1.2.* in c:\python311\lib\site-packages (from platformio) (1.2.0)
Requirement already satisfied: packaging>=17.0 in c:\python311\lib\site-packages (from marshmallow==3.->platformio) (23.1)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\python311\lib\site-packages (from requests==2.->platformio) (3.1.0)
Requirement already satisfied: idna<4,>=2.5 in c:\python311\lib\site-packages (from requests==2.->platformio) (3.4)
Requirement already satisfied: certifi>=2017.4.17 in c:\python311\lib\site-packages (from requests==2.->platformio) (2023.5.7)
Requirement already satisfied: anyio<5,>=3.4.0 in c:\python311\lib\site-packages (from starlette==0.26.->platformio) (3.7.0)
Requirement already satisfied: h11>=0.8 in c:\python311\lib\site-packages (from uvicorn==0.22.->platformio) (0.14.0)
Requirement already satisfied: sniffio>=1.1 in c:\python311\lib\site-packages (from anyio<5,>=3.4.0->starlette==0.26.*->platformio) (1.3.0)
PS C:\Users\Pluim.platformio\penv\Scripts>

ivankravets, 2023-07-05:
> You are using pip version 9.0.1, however version 23.1.2 is available.

Could you try to open system terminal `cmd` and type 
```
pip install -U pip
```

Restart VSCode. Should work now.

closed 2023-07-05 by ivankravets

WPluim, 2023-07-05:
Thanks Ivan. Your suggestion worked like magic!

The last thing I had tried before was: PS C:\Users\Pluim> py -m pip install --upgrade pip

(in Windows PowerShell)

You wrote: Could you try to open system terminal cmd and type: pip install -U pip and: Restart VSCode. Should work now.

The result in the cmd window was: Requirement already satisfied. See PNG file.

After restarting VSCode the message was:

[7/5/2023, 4:12:03 PM] Unable to resolve configuration with compilerPath ***@***.***/bin/xtensa-esp32-elf-gcc.exe".  Using "cl.exe" instead.

But after restarting again all problems have disappeared.

Thanks a lot! Wilfried Pluim