platformio/platformio-core-installer #3017: Installation Manager

Jaramyy, 2023-07-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.79.2
  PIO IDE: v3.2.0
  System: Linux, 5.19.0-46-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/jaramy/.platformio/.cache/tmp/get-platformio-1.1.3.py", line 69, in <module>
    main()
  File "/home/jaramy/.platformio/.cache/tmp/get-platformio-1.1.3.py", line 61, in main
    bootstrap()
  File "/home/jaramy/.platformio/.cache/tmp/get-platformio-1.1.3.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/jaramy/.platformio/.cache/tmp/.piocore-installer-sqolmepn/tmp9j51s32h/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/jaramy/.platformio/penv/bin/python'
 -> Installer version: 1.1.3
Platform: Linux-5.19.0-46-generic-x86_64-with-glibc2.35
Python version: 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/jaramy/.platformio/penv

	at /home/jaramy/.vscode/extensions/platformio.platformio-ide-3.2.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5300
	at ChildProcess.o (/home/jaramy/.vscode/extensions/platformio.platformio-ide-3.2.0-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at Socket.<anonymous> (node:internal/child_process:479:11)
	at Socket.emit (node:events:513:28)
	at Socket.emit (node:domain:489:12)
	at Pipe.<anonymous> (node:net:757:14)
  ```

ivankravets, 2023-07-05:
Could you provide an output of 
```
ls -al /home/jaramy/.platformio/penv/bin
```

Jaramyy, 2023-07-07:
the output is:

ls: cannot access '/home/jaramy/.platformio/penv/bin': No such file or directory

ivankravets, 2023-07-07:
Could you remove `/home/jaramy/.platformio/penv` folder and restart VSCode? Does it work now?

Jaramyy, 2023-07-07:
I remove this folder but it still has the same problem.

ivankravets, 2023-07-07:
Could you contact me directly (me@ikravets.com) and provide a remote session to debug this issue? It would be great to fix it if the problem persists on our side.

Jaramyy, 2023-07-07:
I already sent the email to you.

ivankravets, 2023-07-07:
Resolved in https://github.com/platformio/platformio-core-installer/commit/04010259e57bd30a61045be401a7f0bdd05fab1a

Solution: https://github.com/platformio/platformio-core-installer/issues/85

closed 2023-07-07 by ivankravets