platformio/platformio-core-installer #3018: Installation Manager

Almisuifre, 2023-07-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.79.2
  PIO IDE: v3.2.0
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\get-platformio-1.1.3.py", line 69, in <module>
    main()
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\get-platformio-1.1.3.py", line 61, in main
    bootstrap()
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\get-platformio-1.1.3.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\click\core.py", line 1637, in invoke
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\pioinstaller\core.py", line 70, in install_platformio_core
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\pioinstaller\core.py", line 98, in _install_platformio_core
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\pioinstaller\penv.py", line 71, in create_core_penv
  File "C:\Users\Almi Direct\.platformio\.cache\tmp\.piocore-installer-074ouzxq\tmp_s57zwmg\pioinstaller.zip\pioinstaller\penv.py", line 141, in init_state
  File "C:\Program Files\Inkscape\lib\python3.8\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Program Files\Inkscape\lib\python3.8\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Program Files\Inkscape\lib\python3.8\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Program Files\Inkscape\lib\python3.8\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] Le fichier sp�cifi� est introuvable
 -> Installer version: 1.1.3
Platform: Windows-10
Python version: 3.8.7 (default, Jan 31 2021, 16:39:53)  [GCC 10.2.0 64 bit (AMD64)]
Python path: C:\Program Files\Inkscape\bin\python.exe
Creating a virtual environment at C:\Users\Almi Direct\.platformio\penv

	at c:\Users\Almi Direct\.vscode\extensions\platformio.platformio-ide-3.2.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5300
	at ChildProcess.o (c:\Users\Almi Direct\.vscode\extensions\platformio.platformio-ide-3.2.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

ivankravets, 2023-07-07:
Do you use antivirus tools? See https://github.com/platformio/platformio-vscode-ide/issues/884

Almisuifre, 2023-07-07:
No, no special tools. I think it’s the sequence of installations that corrupted the Python that PlatformIO offers.

To do this, I have virré all configuration files, those of the plafformIO installation. Reinstallation of PlatformIO and Core, and then it redesigned.

if it helps.

closed 2023-07-07 by ivankravets

Almisuifre, 2023-07-07:
I think you can close this ticket that was submitted automatically.

If I have a suggestion, it is a better control of the installation and the management of the core. I can, we may need to make some cohabit and currently it is not easy to find them.

In the meantime, it’s a great soft. I hope it will continue in future years.