platformio/platformio-core-installer #3023: Can't install on Win11. Gives missing venv module error

xcloudx01, 2023-07-10:
I am unable to install PlatformIO Core due to it thinking the venv module is missing, however I can run venv manually without issue. I'm stumped how to get the installation working, any ideas? My Operating System is Windows 11 Home 22H2. It used to work 2 weeks ago, not sure what's changed. Here is the error:

`Error: Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation. Please install this package manually using the OS package manager. For example:  $ apt-get install python3-venv  (MAY require administrator access `sudo`)  -> Installer version: 1.1.3 Platform: Windows-10 Python version: 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)] Python path: C:\Users\cloud\AppData\Local\Programs\Python\Python311\python.exe Creating a virtual environment at C:\Users\cloud\.platformio\penv  	at c:\Users\cloud\.vscode\extensions\platformio.platformio-ide-3.2.0-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:530...`

I've also tried installing it manually following `https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#id2`
which provides the same error:
```
>py get-platformio.py
Installer version: 1.1.3
Platform: Windows-10
Python version: 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)]
Python path: C:\Users\cloud\AppData\Local\Programs\Python\Python311\python.exe
Creating a virtual environment at C:\Users\cloud\.platformio\penv
Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
```
However i can create the venv folder manually with this just fine: `py -m venv C:\Users\cloud\.platformio\penv`
I've tried uninstalling and re-installing the latest Python, and tried installing python3-venv with chocolatey but the issue remains.

xcloudx01, 2023-07-10:
I've stumbled upon a solution that's kinda odd. May give you some info as to what is the cause:

I installed a much older version using the "Install Another Version" button in Visual Studio Code and it has fixed the newer releases installation too. After installing version 2.0.0 and then immediately uninstalling it, I can now install the latest version of PlatformIO just fine. I tried to break it again but i can't, it's working properly now and keeps installing fine, despite my best efforts earlier.

closed 2023-07-10 by xcloudx01

honeydreamssoftwares, 2023-07-18:
I had the same issue on Windows 11. Installing PlatfomIO version v3.1.0 fixed the issue.

SwapnilPande, 2023-07-19:
I also just came across this issue. I had to re-install the Platformio IDE VSCode extension after I upgraded from ubuntu 20 to 22. I received the same error while installing pio core despite having `python3-venv` already installed. I was able to successfully install pio core with v3.2.0, after which I upgraded the extension to v.3.3.0. Looking at release notes for v3.3.0, this could be an isuse related to PIO Core install script 1.2.0.