platformio/platformio-core-installer #3038: PlatformIO: Can not find working Python 3.6+ Interpreter. Please install the latest Python 3 and restart VSCode

schlameel, 2023-07-27:
OS: Linux Mint 20.2
VSCode: 1.80.1
PlatformIO: Current

Installation hung about a third of the way through (according to the progress bar) with the error:
`PlatformIO: Can not find working Python 3.6+ Interpreter. Please install the latest Python 3 and restart VSCode`

I was presented a dialog in the bottom right of the VSCode screen where I could select from four options (something like specify the location of python, search again, quit install, and something else). Specifying the correct location did not work, Search again did not work.  Somewhere I did find a message or note that said look in Help -> Toggle Developer Tools -> maybe console.  In the console I saw a message that install was unable to find `venv`. Although some version of `venv`  IS installed (`virtualenvwrapper` is installed and functional), the `apt` package `python3-venv` is not installed.  Installing with:
`
sudo apt install python3-venv
`
and restarting VSCode allowed the PIO install to complete.

Perhaps there is some way to update the error message or make it easier to find.  The problem with install is clear, `venv` needs to be installed. However, saying that a working python interpreter cannot be found is confusing.

ivankravets, 2024-01-06:
Could someone provide an output from https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#local-download-macos-linux-windows ?

FernandoGarcia, 2024-01-11:
> Could someone provide an output from https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#local-download-macos-linux-windows ?

Hi @ivankravets !

Looks like the problem is related to `python3.7-venv` after install this package the problem was solved.

The output for CURL command was:

```
Installer version: 1.2.1
Platform: Linux-4.15.0-213-generic-x86_64-with-glibc2.26
Python version: 3.7.5 (default, Dec  9 2021, 17:04:37) 
[GCC 8.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/fernandogarcia/.platformio/penv
Error: Can not install PlatformIO Core due to a missed `venv` module in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3.7-venv
```

ivankravets, 2024-01-11:
Yes, duplicate of https://github.com/platformio/platformio-core-installer/issues/85

closed 2024-01-11 by ivankravets