platformio/platformio-core-installer #3044: Installation Manager

pinpc, 2023-08-08:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.81.0
  PIO IDE: v3.3.1
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\zhoup15\.platformio\.cache\tmp\get-platformio-1.2.1.py", line 71, in <module>
    main()
  File "C:\Users\zhoup15\.platformio\.cache\tmp\get-platformio-1.2.1.py", line 63, in main
    bootstrap()
  File "C:\Users\zhoup15\.platformio\.cache\tmp\get-platformio-1.2.1.py", line 49, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\pioinstaller\__main__.py", line 117, in core_check
  File "C:\Users\zhoup15\.platformio\.cache\tmp\.piocore-installer-dchqp07w\tmpa_cm4r01\pioinstaller.zip\pioinstaller\core.py", line 169, in check
  File "C:\Users\zhoup15\AppData\Local\Programs\Python\Python36\lib\subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "C:\Users\zhoup15\AppData\Local\Programs\Python\Python36\lib\subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\zhoup15\AppData\Local\Programs\Python\Python36\lib\subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "C:\Users\zhoup15\AppData\Local\Programs\Python\Python36\lib\subprocess.py", line 1017, in _execute_child
    startupinfo)
OSError: [WinError 1260] This program is blocked by group policy. For more information, contact your system administrator

	at c:\Users\zhoup15\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5300
	at ChildProcess.c (c:\Users\zhoup15\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

pinpc, 2023-08-08:
the platformIO could not installed in the VS Code, please need a help

ivankravets, 2024-01-06:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/3025

closed 2024-01-06 by ivankravets