platformio/platformio-core-installer #3046: Installation Manager

hackerc5, 2023-08-10:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.81.0
  PIO IDE: v3.3.1
  System: Windows_NT, 10.0.22621, x64

  %23 Exception
  ```
  Error:   error: subprocess-exited-with-error
  
  pip subprocess to install build dependencies did not run successfully.
  exit code: 1
  
  [12 lines of output]
  Looking in indexes: https://pypi.doubanio.com/simple
  Collecting setuptools>=40.8.0
    Downloading https://pypi.doubanio.com/packages/c7/42/be1c7bbdd83e1bfb160c94b9cafd8e25efc7400346cf7ccdbdb452c467fa/setuptools-68.0.0-py3-none-any.whl (804 kB)
       ------------------------------------ 804.0/804.0 kB 211.7 kB/s eta 0:00:00
  Collecting wheel
    Downloading https://pypi.doubanio.com/packages/61/86/cc8d1ff2ca31a312a25a708c891cf9facbad4eae493b3872638db6785eb5/wheel-0.40.0-py3-none-any.whl (64 kB)
       -------------------                     32.8/64.5 kB 21.9 kB/s eta 0:00:02
  ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully

hackerc5, 2023-09-16:
您好，我已经收到，尽快给您回复。

ivankravets, 2024-01-06:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/3129

closed 2024-01-06 by ivankravets

hackerc5, 2024-01-06:
您好，我已经收到，尽快给您回复。