platformio/platformio-core-installer #3055: Installation Manager

hxpdeihgu, 2023-08-22:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.81.1
  PIO IDE: v3.3.1
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: ERROR: Could not find a version that satisfies the requirement semantic-version==2.10.* (from platformio) (from versions: 1.0.0, 1.2.0, 2.0.0, 2.1.0, 2.1.1, 2.1.2, 2.2.0, 2.2.1, 2.2.2, 2.3.0, 2.3.1, 2.4.0, 2.4.1, 2.4.2, 2.5.0, 2.6.0, 2.7.0, 2.7.1, 2.8.0, 2.8.1, 2.8.2, 2.8.3, 2.8.4, 2.8.5, 2.9.0)
ERROR: No matching distribution found for semantic-version==2.10.*
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\gh\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'https://github.com/platformio/platformio/archive/develop.zip']' returned non-zero exit status 1.
 -> Installer version: 1.2.1
Platform: Windows-10
Python version: 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)]
Python path: C:\Users\gh\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\gh\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing a development version of PlatformIO Core
Looking in indexes: https://repo.huaweicloud.com/repository/pypi/simple
Collecting https://github.com/platformio/platformio/archive/develop.zip
  Using cached https://github.com/platformio/platformio/archive/develop.zip
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting bottle==0.12.* (from platformio==6.1.11a2)
  Using cached https://repo.huaweicloud.com/repository/pypi/packages/ef/74/39f45da835b8654cfa66ccf6a65cea38d99f940db0a5e0d985de4723a1c9/bottle-0.12.24-py3-none-any.whl (90 kB)
Collecting click<=8.2,>=8.0.4 (from platformio==6.1.11a2)
  Using cached https://repo.huaweicloud.com/repository/pypi/packages/00/2e/d53fa4befbf2cfa713304affc7ca780ce4fc1fd8710527771b58311a3229/click-8.1.7-py3-none-any.whl (97 kB)
Collecting colorama (from platformio==6.1.11a2)
  Using cached https://repo.huaweicloud.com/repository/pypi/packages/77/8b/7550e87b2d308a1b711725dfaddc19c695f8c5fa413c640b2be01662f4e6/colorama-0.4.5-py2.py3-none-any.whl (16 kB)
Collecting marshmallow==3.* (from platformio==6.1.11a2)
  Using cached https://repo.huaweicloud.com/repository/pypi/packages/ae/53/980a20d789029329fdf1546c315f9c92bf862c7f3e7294e3667afcc464f5/marshmallow-3.19.0-py3-none-any.whl (49 kB)
Collecting pyelftools==0.29 (from platformio==6.1.11a2)
  Using cached https://repo.huaweicloud.com/repository/pypi/packages/04/7c/867630e6e6293793f838b31034aa1875e1c3bd8c1ec34a0929a2506f350c/pyelftools-0.29-py2.py3-none-any.whl (174 kB)
Collecting pyserial==3.5.* (from platformio==6.1.11a2)
  Using cached https://repo.huaweicloud.com/repository/pypi/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting requests==2.* (from platformio==6.1.11a2)
  Using cached https://repo.huaweicloud.com/repository/pypi/packages/70/8e/0e2d847013cb52cd35b38c009bb167a1a26b2ce6cd6965bf26b47bc0bf44/requests-2.31.0-py3-none-any.whl (62 kB)
INFO: pip is looking at multiple versions of platformio to determine which version is compatible with other requirements. This could take a while.

	at c:\Users\gh\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5300
	at ChildProcess.c (c:\Users\gh\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1121:16)
	at ChildProcess._handle.onexit (node:internal/child_process:304:5)
  ```

hxpdeihgu, 2023-10-15:
已经收到您的来信，我会尽快回复。胡。

ivankravets, 2024-01-06:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/3129

closed 2024-01-06 by ivankravets

hxpdeihgu, 2024-01-06:
已经收到您的来信，我会尽快回复。胡。

ivankravets, 2024-01-06:
Добре.