platformio/platformio-core-installer #3072: Installation Manager

huozijian, 2023-09-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.82.2
  PIO IDE: v3.3.1
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\lenovo003\.platformio\.cache\tmp\get-platformio-1.2.1.py", line 71, in <module>
    main()
  File "C:\Users\lenovo003\.platformio\.cache\tmp\get-platformio-1.2.1.py", line 63, in main
    bootstrap()
  File "C:\Users\lenovo003\.platformio\.cache\tmp\get-platformio-1.2.1.py", line 49, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\pioinstaller\__main__.py", line 113, in core_check
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\pioinstaller\core.py", line 178, in check
  File "C:\Users\lenovo003\.platformio\.cache\tmp\.piocore-installer-0a_ky9wq\tmpvmlp3axi\pioinstaller.zip\pioinstaller\core.py", line 282, in fetch_python_state
  File "C:\Users\lenovo003\.platformio\python3\Lib\json\__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lenovo003\.platformio\python3\Lib\json\decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 2 column 1 (char 56)
 -> [0x7FFDD02BE3B4] ANOMALY: use of REX.w is meaningless (default operand size is 64)

	at c:\Users\lenovo003\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5300
	at ChildProcess.c (c:\Users\lenovo003\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1091:16)
	at ChildProcess._handle.onexit (node:internal/child_process:302:5)
  ```

huozijian, 2023-09-15:
had try 3 new lenovo Laptops ，had the same issue.

huozijian, 2023-10-15:
这是来自QQ邮箱的假期自动回复邮件。
蕞近莪ぬ忙錒，未螚岌埘囙複伱，so烬請傆諒......

ivankravets, 2024-01-06:
Try please https://github.com/platformio/platformio-core-installer/issues/3111

closed 2024-01-06 by ivankravets