platformio/platformio-core-installer #3119: SSLCertVerificationError

damo1023, 2023-11-29:
Hello,

can not install Platformio Core due to failed certificate verification. Thank you for your help!

regards, Damo

`C:\Temp>python.exe get-platformio.py
Installer version: 1.2.1
Platform: Windows-10
Python version: 3.12.0 (tags/v3.12.0:0fb18b0, Oct  2 2023, 13:03:39) [MSC v.1935 64 bit (AMD64)]
Python path: C:\Users\xxxxxxx\AppData\Local\Programs\Python\Python312\python.exe
Creating a virtual environment at C:\Users\xxxxxx\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/platformio/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/platformio/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/platformio/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/platformio/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))': /simple/platformio/
Could not fetch URL https://pypi.org/simple/platformio/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/platformio/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))) - skipping
ERROR: Could not find a version that satisfies the requirement platformio (from versions: none)
ERROR: No matching distribution found for platformio
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self-signed certificate in certificate chain (_ssl.c:1000)'))) - skipping
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\xxxxx\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.

C:\Temp>

`

ivankravets, 2023-11-29:
Have you solved this issue?

ivankravets, 2024-01-06:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/3129

closed 2024-01-06 by ivankravets