platformio/platformio-core-installer #3122: Installation Manager

HotTabascoSauce, 2023-12-11:


HotTabascoSauce, 2023-12-11:
Error: Traceback (most recent call last): File "C:\Users\charl.platformio.cache\tmp\get-platformio-0.3.5.py", line 69, in main() File "C:\Users\charl.platformio.cache\tmp\get-platformio-0.3.5.py", line 61, in main bootstrap() File "C:\Users\charl.platformio.cache\tmp\get-platformio-0.3.5.py", line 45, in bootstrap import pioinstaller.main File "C:\Users\charl.platformio.cache\tmp.piocore-installer-twf2gaov\tmpr22bqxjo\pioinstaller.zip\pioinstaller__main__.py", line 22, in File "C:\Users\charl.platformio.cache\tmp.piocore-installer-twf2gaov\tmpr22bqxjo\pioinstaller.zip\pioinstaller\core.py", line 25, in File "C:\Users\charl.platformio.cache\tmp.piocore-installer-twf2gaov\tmpr22bqxjo\pioinstaller.zip\pioinstaller\home.py", line 19, in File "C:\Users\charl.platformio.cache\tmp.piocore-installer-twf2gaov\tmpr22bqxjo\pioinstaller.zip\requests__init__.py", line 43, in File "C:\Users\charl.platformio.cache\tmp.piocore-installer-twf2gaov\tmpr22bqxjo\pioinstaller.zip\urllib3__init__.py", line 7, in File "C:\Users\charl.platformio.cache\tmp.piocore-installer-twf2gaov\tmpr22bqxjo\pioinstaller.zip\urllib3\connectionpool.py", line 11, in File "C:\Users\charl.platformio.cache\tmp.piocore-installer-twf2gaov\tmpr22bqxjo\pioinstaller.zip\urllib3\exceptions.py", line 2, in ModuleNotFoundError: No module named 'urllib3.packages.six.moves' Error: spawn C:\Users\charl\AppData\Local\Microsoft\WindowsApps\python.exe ENOENT at C:\Users\charl.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:5004 at s (C:\Users\charl.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:4900) at ChildProcess. (C:\Users\charl.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:4747) at ChildProcess.emit (events.js:223:5) at ChildProcess.cp.emit (C:\Users\charl.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:30:37) at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)

ivankravets, 2024-01-06:
Please update VSCode and PlatformIO IDE extension to the latest versions.

closed 2024-01-06 by ivankravets