platformio/platformio-core-installer #3127: python3 does not seem to update along with Core

classicrocker883, 2023-12-22:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Win11 Pro 22H2
**PlatformIO Version** (`platformio --version`):
 6.1.11
### Description of problem

**~\.platformio\penv\pyvenv.cfg** seems to be outdated. that is  **~\.platformio\python3** uses an older version (probably when PlatformIO is first installed).  
even after updating Core it never changed python3 folder, still showing an older version.

but after deleting that folder, and reloading VSCode, it updated to a more recent python3 version.

#### Steps to Reproduce

1. look in **pyvenv.cfg** file, take note
2. move **python3** directory from **.platformio** somewhere else
3. reload VSCode

### Actual Results
4. look back in **pyvenv.cfg** and you will notice a change, it is updated

### Expected Results
after updating Core, or whether manually or automatically, you'd expect it to update **python3** automatically as well.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

all I am is saying is I would expect **python3** to be updated with having everything else updated. all without having to manually do this, since no one would know this, because I see that **python3** folder hasnt changed all year.

ivankravets, 2023-12-22:
The latest version published in the registry is 3.11.6 https://registry.platformio.org/tools/platformio/python-portable

We will update to the 3.11.7

Please do the following:

1. Close VSCode
2. Remove `C:/Users/***/python3`
3. Remove `C:/Users/***/penv`

Start VSCode.

ivankravets, 2024-01-06:
Please do the following:

    Close VSCode
    Remove C:/Users/***/python3
    Remove C:/Users/***/penv

closed 2024-01-06 by ivankravets

tylerwowen, 2024-03-01:
@ivankravets 

Looks like the configuration below will cause the darwin_x86_64 version of python installed on darwin_arm64 devices. Consequently, the tool chain used for building will be in the x86_64 flavor, and can cause many subsequent issues, such as https://github.com/lvgl/lv_platformio/issues/58. 

```json
                {
                    "name": "python-portable-darwin_x86_64-1.31107.0.tar.gz",
                    "size": 19514642,
                    "checksum": {
                        "sha256": "384387fc8554c9e649dd101b972fee7ea1333fa1b7ff2db6e642d11afd409495"
                    },
                    "system": [
                        "darwin_x86_64",
                        "darwin_arm64"
                    ],
                    "download_url": "https://dl.registry.platformio.org/download/platformio/tool/python-portable/1.31107.0/python-portable-darwin_x86_64-1.31107.0.tar.gz"
                },
```