platformio/platformio-core-installer #3129: Installation Manager:Could not find a version that satisfies the requirement platformio

wcpsoft, 2023-12-29:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.85.1
  PIO IDE: v3.3.2
  System: Linux, 6.5.0-14-generic, x64

  %23 Exception
  ```
  Error: ERROR: Could not find a version that satisfies the requirement platformio (from versions: none)
ERROR: No matching distribution found for platformio
Error: Could not install PlatformIO Core: Command '['/home/rxzhang/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.2.1
Platform: Linux-6.5.0-14-generic-x86_64-with-glibc2.35
Python version: 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/rxzhang/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core

	at /home/rxzhang/.vscode/extensions/platformio.platformio-ide-3.3.2-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5300
	at ChildProcess.c (/home/rxzhang/.vscode/extensions/platformio.platformio-ide-3.3.2-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4598)
	at ChildProcess.emit (node:events:513:28)
	at ChildProcess.emit (node:domain:489:12)
	at maybeClose (node:internal/child_process:1091:16)
	at Socket.<anonymous> (node:internal/child_process:449:11)
	at Socket.emit (node:events:513:28)
	at Socket.emit (node:domain:489:12)
	at Pipe.<anonymous> (node:net:322:12)
  ```

wcpsoft, 2023-12-29:
auto download make don't work. /home/rxzhang/.platformio/penv/bin/pip install platformio is working

ivankravets, 2024-01-06:
Could you remove `/home/rxzhang/.platformio/penv` and restart VSCode? Do you still see this issue?