platformio/platformio-core-installer #3204: Installation Manager in VSCODE Failed to install PlatformIO IDE

nve-sak, 2024-05-19:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.88.1
  PIO IDE: v3.3.3
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\sak\.platformio\.cache\tmp\get-platformio-1.2.2.py", line 71, in <module>
    main()
  File "C:\Users\sak\.platformio\.cache\tmp\get-platformio-1.2.2.py", line 63, in main
    bootstrap()
  File "C:\Users\sak\.platformio\.cache\tmp\get-platformio-1.2.2.py", line 49, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\pioinstaller\__main__.py", line 113, in core_check
  File "C:\Users\sak\.platformio\.cache\tmp\.piocore-installer-acqk6gvk\tmpa_0mqf2s\pioinstaller.zip\pioinstaller\core.py", line 172, in check
  File "C:\PythonNVE\Lib\subprocess.py", line 465, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\PythonNVE\Lib\subprocess.py", line 546, in run
    with Popen(*popenargs, **kwargs) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\PythonNVE\Lib\subprocess.py", line 1022, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\PythonNVE\Lib\subprocess.py", line 1491, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [WinError 1260] Dette programmet er blokkert for gruppepolicy. Kontakt systemansvarlig for mer informasjon

    at c:\Users\sak\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5323
    at ChildProcess.i (c:\Users\sak\.vscode\extensions\platformio.platformio-ide-3.3.3-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4621)
    at ChildProcess.emit (node:events:517:28)
    at ChildProcess.emit (node:domain:489:12)
    at maybeClose (node:internal/child_process:1098:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:303:5)
  ```

kana542, 2024-08-26:
the error code (WinError 1260) indicates that the execution of the program was blocked due to a restriction set by Group Policy. I recommend checking your local Group Policy settings. additionally, you might want to check if any software, such as security or antivirus programs, is blocking the execution

eabase, 2024-11-27:
@kana542 
If the program is blocked by windows security is probably for a very good reason!
Why don't you tell us what the code is trying to do instead, and why this is not explained **anywhere**!?

kana542, 2024-12-17:
@eabase
did you go through the code and understood what it was trying to do? 
or where did this conclusion come from?

yes maybe, or then maybe not.
things like these happen usually cuz of antivirus/security/hardening software that have enforced a more strict security policies.
its good thing and also bad thing, since it causes issues like this, unless you whitelist the process.

as in this case, the issue comes from part of get-platformio-1.2.2.py, that attempts to set up or verify the platformio core environment, since it uses standard python libraries/subprocess to execute commands necessary for platformio's installation.

closed 2025-04-26 by stale[bot]