platformio/platformio-core-installer #3268: Unable to Install on Mac

DaveL17, 2024-12-10:
### Configuration

**Operating system**:
macOs Sequoia 15.1.1
Macbook Pro 2020 Intel

**PlatformIO Version** (`platformio --version`):
3.3.3

### Description of problem
After repeated attempts, I was unable to install PIO within VSCode or PyCharm. I'm no longer able to find the precise error message text, but the crux of the issue was a `Permission Denied [13]` issue related to the `/Users/User/Library/Caches/pip/wheels/5c` folder. I initially chalked this up to a Python version issue since macOS defaults to Python 2.7, then I thought that it might be related to the increased security of Sequoia (which it could still potentially be).

### Steps to overcome problem
1. Manually change the permissions of the `/Users/User/Library/Caches/pip/wheels/` folder from `Read Only` to `Read & Write` for the subject user.
2. Manually create the `/Users/User/Library/Caches/pip/wheels/5c` folder.
3. Ensure `/Users/User/Library/Caches/pip/wheels/5c` folder is also `Read & Write`.
4. PIO should successfully install.

ivankravets, 2024-12-10:
> Permission Denied [13]

This is a system configuration issue and isn't related to the PlatformIO.

Could you try to restart your macOS?

closed 2024-12-10 by ivankravets

DaveL17, 2024-12-10:
Restarting didn't help. I found it odd that on each attempt, it always tried to write to the same cache file location and that I have had no issues installing other packages with pip. I raised it primarily as an FYI as it may come up for other Mac users on Sequoia, and the workaround is less than obvious.  Cheers.